ra3xdh/qucs_s #298: log/ln issue in Nutmeg equation

jlinkels, 2023-08-12:
Hello,

As far as I know I use Ngspice as simulator.
This is what the simulation console says:
```
Ngspice started...
No compatibility mode selected!
```
I have entered a Nutmeg equation:
`20*log(abs(v_out))`
But that shows the value of the natural logarithm.

Eventually I tried:
`20*log(abs(v_out))`
And that shows the correct value.

In the online documentation I find that Ngspice functions include log() and ln(). SPICE OPUS does use log() and log10().

Is this the documentation which is lagging?

Help -> About shows that I am using QUC-S version 1.1.0 (unknown). But (C) 2011-2016. So I am not 100% sure I am using the latest version. I am running Debian 11 Bullseye and I assume I installed it from the Debian repository.

ra3xdh, 2023-08-13:
Qucs-S doesn't apply any conversion for Nutmeg equations. These equations are translated to the netlist as is. Qucs-S also doesn't define new functions related to logarithms. Please ask this question at the Ngspice forum, because it is not related to the GUI level. 

> QUC-S version 1.1.0 (unknown). But (C) 2011-2016. So I am not 100% sure I am using the latest version.

Yes, the `v1.1.0` is the latest version. There is a typo in the About dialog.

jlinkels, 2023-08-13:
@ra3xdh ra3xdh Thanks for the explanation. I did not realize that.

closed 2023-08-13 by jlinkels