ra3xdh/qucs_s #302: Arbitrary complex load

0range0range, 2023-08-21:
Hello and thanks for the great software.

**Background**: I want to test output impedance of a common base amplifier with a resonance circuit as a load. I am not sure whether just measuring S22 is the correct way of doing this.

One of the experimental ways would be to use fixed frequency signal source and adjustable load at the output, until the power transfer is maximum. Then conjugate to that load would be output impedance.
 
Therefore, I tried to use complex numbers directly like this:
[image]

**My plan is to make a parametrized series RL, RC circuits, but is there a direct way to achieve this somehow?**

In a post here ( http://qucsstudio.de/forums/topic/complex-impedance/ ) “Frequency Domain” component was mentioned, which I could not find in Qucs-S. 

Also, any other suggestions and best practices on how to get the output impedance in simulation is more than welcome. Maybe I am approaching it incorrectly altogether...

Thanks!

ra3xdh, 2023-08-21:
Ngspice which is used as the default simulation kernel for Qucs-S doesn't support definition of the resistance as the complex number. You have to compose the load as the RC or RL parametric circuit. See the section 3.3 of the Ngspice reference manual for allowed expression syntax for expression-defined elementary devices. https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml  

> In a post here ( http://qucsstudio.de/forums/topic/complex-impedance/ ) “Frequency Domain” component was mentioned

QucsStudio is not related to Qucs-S project. It is maintained by another team and not open source software. QucsStudio uses another simulation kernel and not compatible with SPICE.

0range0range, 2023-08-22:
I think I understood how to do it. I made it with an inductor and it works for positive values of reactance. But negative values with capacitor do not work. I get 'nan' values on the smith chart. 
[image]

I also tried to simulate RC separately, but same problem. Irrespective of whether Zim is positive or negative.
[image]

**Why it does not work with capacitor, but works with inductor?**

tomhajjar, 2023-08-30:
[image]

[Complex_Z_ngspice.zip](https://github.com/ra3xdh/qucs_s/files/12476501/Complex_Z_ngspice.zip)

0range0range, 2023-09-15:
@tomhajjar Thanks a lot for your help. Sorry for the long responce. I just do not understand how can I reuse it as a subcircuit in another place.

My idea was that I would just enter two parameters (Re and Im parts), whenever I need to substitute a complex load. Like capacitances for transistor model for example.

tomhajjar, 2023-09-15:
Here are two examples of parameter passing and parameter sweeping. 

In your case, you make a subcircuit of your RLC network for the complex load and if you want to alter the "default" values, use parameter passing. 

[Parameter.zip](https://github.com/ra3xdh/qucs_s/files/12620442/Parameter.zip)

ra3xdh, 2023-10-04:
@0range0range I have tried to reproduce your problem and found a solution. It was required to use `C={...}` syntax for capacitor expression. Otherwise it gives `nan`. It was not obvious from Ngspice manual. Look at the attached schematic for solution: 
[image]
[image]

ra3xdh, 2023-10-06:
No further interaction from issue reporter. Assume the problem was resolved after fixing syntax. Closing ticket as completed.

closed 2023-10-06 by ra3xdh

0range0range, 2023-10-06:
@ra3xdh This worked for me as well. Thanks a lot to everyone involved! 

I would have never tried C=C=...  . Should I inform ngspice developers somehow? Or is it intended behavior?

ra3xdh, 2023-10-06:
> Should I inform ngspice developers somehow? 

I think this syntax is used by Ngspice intentionally to distinguish model and expression as capacitor parameter.