ra3xdh/qucs_s #303: Build issue with VERSION file on macOS

hugolundin, 2023-08-22:
I had issues building the project on macOS (Clang 14.0.3, x86_64), repeatedly getting this error (> 20 times): 

```shell
...
qucs_s/version:1:1: error: expected unqualified-id
2.0.0
...
```

Some investigation showed that a similar issue has up in other projects (eg. [openMVG](https://github.com/openMVG/openMVG/commit/1c5fe8166c5bb463e1e9487623937dbea1823377) and [UIUC-PLL/charm](https://github.com/UIUC-PPL/charm/issues/2844)) and that the culprit was the `VERSION` file, causing problems due to macOS generally being case-insensitive. Renaming the file from `VERSION` to `VERSION.txt` solved the issue. 

Different projects have chosen different paths for getting rid of the issue (for example making the `VERSION` file C++ compilable, replacing `VERSION` with a header file or renaming it to, for example, `QUCS_S_VERSION`). I don't know what the right approach is for QUCS-S, but wanted to highlight the issue in an effort to make it viable to compile and run on macOS.

ra3xdh, 2023-08-22:
The solution may be to exclude the source root directory containing `VERSION` file from the include path. It's need to find a way how to do it in cmake.

ra3xdh, 2023-08-22:
The path containing the VERSION file was included by `converter` utility that was added recently. Now the include files conflict should be resolved. Please try to build the latest state of the `current` branch and report the result.

hugolundin, 2023-08-23:
That works much better. Thanks!

closed 2023-08-23 by hugolundin