ra3xdh/qucs_s #304: Blurry icons on macOS

hugolundin, 2023-08-22:
Icons are blurry when using macOS with High DPI (Retina). See the attached image. 

[image]

It seems like Qt should have means for solving this: https://doc.qt.io/qt-6/highdpi.html

ra3xdh, 2023-08-22:
Yes, I am aware about issues with icons. The same appears using KDE on Linux and display scaling. 

[image]

The icons in the Qucs-S are designed as fixed resolution 22x22 PNG bitmaps. These icons still be inherited from Qt3 version. And this will look poor when using HiDPI resolution. The only solution for this issue is to redesign **all** button icons located in `qucs/bitmaps`. The usage of the SVG icons would be also good. But this task cannot be done in the near future because of need to redraw **all** icons. I am programmer, but not a graphical designer and cannot perform this task by myself. The contribution is highly welcome here. 

Summarizing all above, don't expect the icons to be fixed in the near years, unless someone will redesign the icons and propose a PR.

ra3xdh, 2023-10-15:
I have started to replace toolbar icons by standard SVG icons from KDE Oxagen theme. This icon set should look good for both dark and light desktop themes. The most of toolbar icons could be replaced by standard icons. But 9 icons should be designed from scratch. Probably Oxygen icons could be used as template. I have uploaded updated icons to the `svg_icons` branch: https://github.com/ra3xdh/qucs_s/tree/svg_icons 

I don't consider the redesign of components icons because a huge amount of work. 

[image]

ra3xdh, 2023-10-16:
I have finished update of the toolbar icons. The result could be seen on the screenshot. I am planning to include updated icons in the next release. The following tasks are remaining:

* It's need to implement automatic generation of the component icons. This will allow to avoid redesign of more than 100 icons. 
* The diagram icons cannot be generated automatically. These 11 icons should be redrawn by hand.

ra3xdh, 2023-10-16:
Here is the new toolbar look:
[image]

ra3xdh, 2023-10-16:
See #323

ra3xdh, 2023-10-18:
See #332 for diagram and paintings icons

ra3xdh, 2023-10-19:
I have updated diagram icons. Here is the new application look. It is not possible to update remaining 250+ device icons manually. Automatic icons generator need to be implemented. 
[image]

ra3xdh, 2023-10-20:
See #334 for icons rendering. Here is the new application look.
[image]

ra3xdh, 2023-10-20:
New auto-generated icons after adjustment.
[image]

hugolundin, 2023-10-20:
Amazing work!

ra3xdh, 2023-10-21:
Resolved with #332 and #334. New icons will be available since the next release. I have no possibility to check everything with 4k resolution hardware, but everything works fine using fullHD and scaling applied.

closed 2023-10-21 by ra3xdh

hugolundin, 2023-10-22:
I can do a sanity check on my retina Mac later next week. But this seems fantastic, great job!

hugolundin, 2023-10-23:
[image]

The icons look great on my Macbook Air (with scaled resolution) now! Thank you!

ra3xdh, 2023-10-24:
@hugolundin Thanks for testing on MacOS! I am planning to prepare the release packages by the end of this week.

PR #323 (2023-10-17): Svg icons
Toolbar icons are replaced by SVG images. Partially resolves #304
PR #334 (2023-10-21): Render icons
This PR implements dynamic rendering of the icons. Fully fixes #304