ra3xdh/qucs_s #306: Tracking qucs-s in the packaging ecosystem

luzpaz, 2023-08-31:
**Current snapshot of qucs-s in downstream package repos**:

[image]

**Here are some attempts to update or implement qucs-s downstream**:

- [ ] Chocolatey (windows)
- [ ] Debian
- [ ] Gentoo
- [X] nixpkgs https://github.com/NixOS/nixpkgs/issues/250576#event-10242240167
- [ ] OpenSUSE
- [ ] Ubuntu
- [ ] winget (windows)
- [ ] wikipedia (create a page for qucs-s, any volunteers?)

luzpaz, 2023-08-31:
Added link to previously opened Gentoo new package request: https://bugs.gentoo.org/891303

ra3xdh, 2023-08-31:
> Added link to previously opened Gentoo new package request:

The Gentoo member has asked if the Qucs-S is ported to Qt6. Yes, the porting has been finished in 2.0.0. See #190. @luzpaz , please answer about this at the Gentoo forum.

closed 2024-02-28 by zergud

luzpaz, 2024-02-28:
bumped gentoo ticket again

PR #619 (2024-02-28): Add tiny repository tracking badge to README
Hi! 

This PR offers fancy stuff — a little badge to README which when clicked leads to the page tracking Qucs-S in various repositories. I think it would be nice to have it among the badges already in README.

Also merging this may be considered the reason to close #306 as the badge actually tracks Qucs-S in repos.