ra3xdh/qucs_s #307: Can't find model files with ngspice

dwarning, 2023-08-31:
After open example file BSIM4_Ring_Osc.sch from ngspice example directory and simulate I got:
```
Ngspice started...
Error: Could not find include file nmos4p0.mod
Note: No ".plot", ".print", or ".fourier" lines; no simulations run
Simulation finished
```
Both include files are in the folder. Also given the entire path doesn't help.

ra3xdh, 2023-09-01:
I have just checked this schematic and everything works as expected after the setting of the full path to modelcard files. You doing something wrong. I assume you have forgot to save the subcircuit after the correction of the paths.

dwarning, 2023-09-01:
Simple to say I will do something wrong!
The long path was only an additional experiment. In the provided BSIM4_inv.sch stands only "nmos4p0.mod" and "pmos4p0.mod". And the generated netlist have correct .include lines. I can simulate this file with ngspice very well on command line.
Maybe I have to set a working directory somewhere? Forgot to say I am under Ubuntu 22.04.

ra3xdh, 2023-09-01:
I still cannot reproduce this issue. The schematic file is definitely not broken. I am using Ubuntu22.04 and Ngspice-41. I have also tested Ngspice-39. Everything works as expected. Here are my simulator settings and screenshot from the application:
[image]
[image]

dwarning, 2023-09-01:
This are my settings and the error message:
[image]
[image]
qucs-s is build by latest github source.
Same for ngspice - but also tried with older version.
BTW: I see it is the onlyest version with external models.

ra3xdh, 2023-09-01:
Sorry, but I still cannot reproduce this.

ra3xdh, 2023-09-01:
I have just tested Windows version and I cannot reproduce the problem with .INCLUDE. Everything works as expected on Windows platform too.

dwarning, 2023-09-01:
OK, perhaps my system is wrong.
I will close it.
Thanks

dwarning, 2023-11-09:
After installing latest linux and windows version I have still this problem, that  the bsim4 example has the problem of not finding nmos4.mod and pmos4.mod.
All examples without include or which have the models included in the schematic are working fine.
I can simulate on commandline the netlist.cir with my ngspice41+ without any problem.

ra3xdh, 2023-11-10:
Sorry, I cannot reproduce this error. I performed the following steps:

* Install Ubunutu-22.04 into the fresh VirtialBox. 
* Install qucs-s v2.1.0 package from here: https://software.opensuse.org/download.html?project=home%3Ara3xdh&package=qucs-s
* Compile and install ngspice-41 into the `/opt/ngspice-41` prefix
* Specify the ngspice path in the Qucs-S settings
* Copy qucs-s examples into the `$HOME/examples` directory
* Open BSIM4_Ring_Osc.sch file. 
* Fix paths to `*.mod` files in BSIM4_inv.sch file
* Save BSIM4_inv.sch file
* Launch simulation with F2
* Everything works as expected (See the screenshot)
[image]

Please double check that you have performed everything mentioned above. The simulation also works fine using the system package Ngspice-36 from the repository.

dwarning, 2023-11-10:
Thanks to spend the effort.
I will perform same procedure soon. But what do you meant with "Fix paths to *.mod files in BSIM4_inv.sch file"?
The both files reside in same directory as all the other BSIM4 example files. Can you attach this small file, please.

ra3xdh, 2023-11-10:
I meant to specify the absolute path to the model files. Relative path will not work. The schematic should look as the following. The exact path will depend on the location of the BSIM4 directory. Don't forget to save the BSIM4_inv.sch after fixing the paths. Otherwise the netlist will be not updated.
[image]

dwarning, 2023-11-10:
Vadim,
now it works: Absolute path is mandatory.
I thought files are in the dir and can be used. But it seems not the case. File=nmos4p0.mod seems a special case of relative path.
Thank you!

closed 2023-11-10 by dwarning