ra3xdh/qucs_s #308: Xanalogue Library Graphics

tomhajjar, 2023-09-19:
Four Xanalogue Library devices display garbage for the device graphic that is in the middle of the schematic symbol outline. 

However if I cut and paste the symbol info into a schematic, the graphic is correct.

[image]
[image]
[Symbol_creation_prj.zip](https://github.com/ra3xdh/qucs_s/files/12655529/Symbol_creation_prj.zip)

ra3xdh, 2023-09-19:
I cannot reproduce the issue with symbols using Linux platform. It seems Windows-only issue. Need to make more debugging to fix this.
[image]

tomhajjar, 2023-09-22:
I have some more info. 

The Xanalogue.lib component XA_CSWO uses "~" but the "Show model" prevue displays garbage instead of "~".

The Xanalogue.lib components XA_int_DE and XA_int_SE use "?" instead of an integral symbol. The "Show model" prevue displays garbage instead of "?". 

I'm fixing the graphics issues by using Arc's and Lines instead of "~" and "?". I'm also making the graphics consistent and cleaner. It will be a few days before I'm done.

[image]

[image]

ra3xdh, 2023-09-22:
@tomhajjar This issue should be fixed in C++ level. I guess the unicode support should be switched somewhere explicitly for Windows. @zergud  will provide debugging in the next week.

tomhajjar, 2023-09-25:
Still puzzled how an ASCII "?" can display an integral symbol...

ra3xdh, 2023-09-27:
Fixed by #316. Closing as completed. The correction in the library is not needed. The fix will be available since the next release.

commit 5a964277 (2023-10-26): set encoding for qtextcodec
fixes #308
closed 2023-09-27 by ra3xdh