ra3xdh/qucs_s #310: Add low voltage for the rectangle sources

ljbade, 2023-09-21:
The rectangle voltage source doesn't provide a way to set the low voltage, only the high voltage.

I wanted a square wave with a negative low voltage. I think I can work around it by using another DC voltage source.

ra3xdh, 2023-09-21:
I cannot change the existing source because all schematics containing it will be broken. You can use the full SPICE definition with arbitrary source (V-source) as the workaround. The example is shown below. See the Ngspice manual for description https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml#magicparlabel-4121
[image]

ra3xdh, 2023-09-23:
I have added a fix for this. The backward compatibility of the square wave source should not be broken. Since the next version a new parameter `U0` representing the low voltage will be added. Qucsator doesn't support the low voltage of the square wave source, so this new parameter will be SPICE-only (supported by Ngspice/XYCE). Qucsator will ignore it and substitute zero. This fix will be available since the next release. Closing ticket as completed. 
 
[image]

closed 2023-09-23 by ra3xdh