ra3xdh/qucs_s #328: Switch versioning to CalVer

ra3xdh, 2023-10-18:
Qucs-S uses semantic versioning where the first digit means major version, the second digit minor version, and the third version patches. Looking at the recent releases, it could  seen that the last digit is almost not in use. For example the 2.0.0 follows right after 1.1.0. It also may be hard to distinguish major and minor releases. For example the next version should be major release because of tuner and s2p files support. The project would benefit after switching to CalVer. 

I propose the following system: `YY.MM.XX` for example `23.10.0`

* The first digit means Year of the release
* The second digit means Month of the release
* The last digit means patches within month

The backward compatibility of schematic files should not be broken after switching to CalVer.

ra3xdh, 2023-10-26:
The first release of the next year will be switched to CalVer. The next milestone set to 24.1.0

closed 2023-10-26 by ra3xdh

mid-kid, 2024-02-19:
I'm not sure how this makes it easier to distinguish major and minor releases. In semver, the major version number is used to distinguish *backwards-incompatible* changes, be that either because of file formats or incompatible changes in the user-facing workflow (the user has to relearn the interface). There's nothing wrong with practically never incrementing the major version number.