ra3xdh/qucs_s #333: install QUCS-S with ngspice41 under UBUNTU 22.04

olisnr, 2023-10-20:
i need a newer ngpice version because of verilog-a models. i installed the version 4.1 from source. but now i cant install QUCS-S, because it wants to overwrite some files.

the thing is, before installing ngspice41 i was able to install QUCS-S 2.0.0.1. but i was only able to install ngspice41 after removing QUCS-S.

i dont understand whats going on:

`ich@t14:~$ sudo apt install qucs-s
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
qucs-s is already the newest version (2.0.0-1).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 qucs-s : Depends: ngspice but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
ich@t14:~$ sudo apt --fix-broken install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  ngspice
Suggested packages:
  ngspice-doc
The following NEW packages will be installed:
  ngspice
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
Need to get 0 B/2’353 kB of archives.
After this operation, 7’972 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
(Reading database ... 676307 files and directories currently installed.)
Preparing to unpack .../ngspice_36+ds-1_amd64.deb ...
Unpacking ngspice (36+ds-1) ...
dpkg: error processing archive /var/cache/apt/archives/ngspice_36+ds-1_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/ngspice/analog.cm', which is also in package libngspice-kicad 0.1-38~202211100131+14f42f805~ubuntu22.04.1
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/ngspice_36+ds-1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
ich@t14:~$ 
`

olisnr, 2023-10-20:
the formating was bad. i try again

```
ich@t14:~$ sudo apt install qucs-s
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
qucs-s is already the newest version (2.0.0-1).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 qucs-s : Depends: ngspice but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
ich@t14:~$ sudo apt --fix-broken install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  ngspice
Suggested packages:
  ngspice-doc
The following NEW packages will be installed:
  ngspice
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
Need to get 0 B/2’353 kB of archives.
After this operation, 7’972 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
(Reading database ... 676307 files and directories currently installed.)
Preparing to unpack .../ngspice_36+ds-1_amd64.deb ...
Unpacking ngspice (36+ds-1) ...
dpkg: error processing archive /var/cache/apt/archives/ngspice_36+ds-1_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/ngspice/analog.cm', which is also in package libngspice-kicad 0.1-38~202211100131+14f42f805~ubuntu22.04.1
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/ngspice_36+ds-1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
ich@t14:~$ 
```

ra3xdh, 2023-10-20:
It is not a bug in Qucs-S. You have a broken dependency, because you have installed ngspice to default prefix `/usr/`. Now Ngspice-41 is conflicting with system package. Be careful when installing something with `make install`. You should alway specifiy prefix in `/opt/` or another location to avoid such conflict. Use the following configure for Ngspice:

~~~
configure --prefix=/opt/ngspice-41 <other_options>
~~~ 

This will install Ngspice-41 to `/opt`

ra3xdh, 2023-10-20:
Also try to remove `libngspice-kicad` This may be package conflict in Ubuntu.

ra3xdh, 2023-10-20:
Ubuntu doesn't provide `libngspice-kicad` in system repository. Where did you take it? It may have conflicts.

olisnr, 2023-10-23:
i installed ngspice41 from source. and in basically i would like it to be the standard ngpsice, because of the OSDi models. i would prefere to install QUCS-S also from source, so that it dont have to install old versions. or i could try the flatpak... may be i find a soluiton...

olisnr, 2023-10-23:
building from source is working with ngspice41. thaks for Your help!

ra3xdh, 2023-10-29:
I did more investigation on this. The `ngspice-kicad` from KiCAD PPA conflicts with `ngspice` from system repository. The `ngspice-kicad` provides only shared library and cannot serve as the replacement for system `ngspice`. I have informed PPA maintainer about this issue. Waiting for reaction.

The following workaround is proposed:

* I have moved `ngspice` package to the recommended packages. At least this will allow to install Qucs-S if KiCAD PPA is enabled and avoid building Qucs-S from source. 
* The `ngspice-kicad` package is not sufficient for Qucs-S. The simulation will not work with it. The Ngspice is required to be build from source and installed in `/opt`
* It is possible to switch KiCAD version from PPA to system repository. The KiCAD package from system repository has no conflicts with `ngspice`

ra3xdh, 2023-10-29:
After some experiments the following workaround is recommended:

* remove `ngspice-kicad` package
* install `qucs-s` , `ngspice` , and `libngspice0`
* simulation will work in both KiCAD and Qucs-S+Ngspice

You may require to compile Ngspice from source if the version higher that repository package is required.

olisnr, 2023-10-30:
yes, compiling ngspice from source is the best way

ra3xdh, 2023-11-08:
See https://gitlab.com/kicad/packaging/kicad-ubuntu-builder/ngspice-kicad-pkg/-/issues/4

ra3xdh, 2023-12-02:
I cannot provide the full fix for this issue on Qucs-S package level. Need to wait until ngspice-kicad maintainer will resolve the conflict with the system package. Still be no reaction from KiCAD team. Closing for now.

closed 2023-12-02 by ra3xdh