ra3xdh/qucs_s #342: Library Creation Issues

tomhajjar, 2023-10-28:
I created a library called test.lib made up of two devices, a generic OpAmp and a delay element for ideal.lib. 

Component "Qucs-S Delay model" seems OK.

Component "genopa_sub" will not function if I don't send the model genericopa1.cir with the library. How do I make a self contained library independent of external files pointed to by the following:

Test_genopa_sub.sch
.INCLUDE "C:/Users/Tom Hajjar/.qucs - S/Ideal Library Test_prj\genericopa1.cir"

[Test_Lib_prj.zip](https://github.com/ra3xdh/qucs_s/files/13194410/Test_Lib_prj.zip)

ra3xdh, 2023-10-28:
This question was asked several times before. See also #140 You have to use `Spice netlist` device instead of `SpiceLibComp` if you are intending to create libraries from subcircuit. The purpose of SpiceLibComp is to use *unmodified* SPICE libraries with Qucs-S. That's why this device it still referencing external library.

ra3xdh, 2023-10-30:
Assume the question about difference between *SPICE file* and *Spice Lib Comp* is closed.

ra3xdh, 2023-10-30:
Closing this.

closed 2023-10-30 by ra3xdh