ra3xdh/qucs_s #343: Add system library of Verilog-A defined devices

ra3xdh, 2023-10-28:
The microstrip devices models #94 will be most likely implemented in form of Verilog-A modules. These modules should be compiled with OpenVAF and attached to Ngspice. It's need to provide a way to attach Verilog-A model to system devices. I propose the following approach:

* Put a set of VA models in `share/qucs-s/va_library` directory in the installation root. 

The two ways to integrate these models are possible:

* Add a button named like *Compile system libraries* in the application setting dialog near the OpenVAF configuration input field. Compiled OSDI modules will be placed to the `$HOME/.qucs/osdi_modules` location and attached automatically while forming `.spiceinit` config. Modules should be build once.

* Scan schematic recursively and compile OSDI modules automatically before the simulation if some device referencing Verilog-A model is found.

The combination of these two approaches is also possible. This feature should be added in order to implement #94 in the next release.

zergud, 2023-10-30:
there is a more flexible solution for library location:
- SYSTEM - located in share/qucs-s/va_library (readonly)
- USER - located in user home dir $HOME/.qucs/va_library
- PROJECT - located in project folder

USER and PROJECT directory are writeble for user

all compiled OSDI modules will be placed to the $HOME/.qucs/osdi_modules location and attached automatically while forming .spiceinit config

compilation approach, control duplicated name and recompilation condition (e.g. via file timestamp) are subject to discuss

ra3xdh, 2024-09-10:
The development of OpenVAF compiler still be suspended. No more visible contributions since October 2023. I would close this task, because the future of OpenVAF still be not clear. May be reopened if some maintainer will appear in OpenVAF project.

zergud, 2024-09-10:
https://openvaf.semimod.de/blog/vacask/
"some maintainer" appear :)

ra3xdh, 2024-09-10:
Maybe it makes sense to switch to openvaf-reloaded fork and use it by default.

tomhajjar, 2024-09-10:
Windows builds not provided...

Kreijstal, 2024-11-26:
> Windows builds not provided...

OpenVaf compiles with msys2
https://github.com/Kreijstal/OpenVAF/actions/runs/12016195403/job/33495922743
But it doesn't compile with clang yet

tomhajjar, 2024-11-26:
I reloaded all the MS tools so I can compile va files using OpenVAF once more.

As stated in link below, some va files cannot be compiled inside Qucs-S so the Dietmar/Holger created batch file is used. I bodged the bat file to work on my PC.

Will this new branch of OpenVAF have all the same functions as the original?

https://github.com/ra3xdh/qucs_s/discussions/1100

tomhajjar, 2024-11-26:
You might want to try Dietmar's batch file and see if all the va files in the code folder compile. You might have to alter paths in the batch file to reflect you system or use a folder structure like Dietmar.

https://github.com/dwarning/VA-Models

Kreijstal, 2024-12-01:
```
(venv) kreijstal@kreijstalnuc:~/git/VA-Models/code$ ./openvaf-compile-va.sh
Finished building angelov.va in 0.18s
Finished building angelov_gan.va in 0.20s
Finished building asmhemt.va in 4.15s
Finished building bsim4.va in 2.73s
Finished building BSIM6.1.1.va in 1.53s
Finished building bsimbulk106.va in 1.77s
Finished building bsimbulk.va in 2.30s
Finished building bsimcmg.va in 1.57s
Finished building bsimcmg.va in 3.76s
Finished building bsimimg.va in 1.34s
Finished building bsimsoi.va in 1.42s
Finished building diode_cmc.va in 0.66s
Finished building ekv26.va in 0.18s
Finished building ekv3.va in 0.54s
Finished building epfl_hemt.va in 0.11s
Finished building fbh_hbt-2_1.va in 0.16s
Finished building fbh_hbt-2_3.va in 0.23s
Finished building hicumL0_v2p1p0.va in 0.31s
Finished building hicumL2.va in 0.61s
Finished building hicumL2V3p0p0.va in 0.57s
Finished building hisim2.va in 3.27s
Finished building hisimhv.va in 6.45s
Finished building hisimsoi.va in 2.17s
Finished building hisimsotb.va in 1.34s
Finished building igbt3.va in 0.09s
Finished building L_UTSOI_102.va in 1.99s
Finished building bjt504.va in 0.28s
Finished building bjt504t.va in 0.39s
Finished building bjt505.va in 0.34s
Finished building bjt505t.va in 0.47s
Finished building mosvar.va in 0.27s
Finished building mvsg_cmc_3.2.0.va in 1.47s
Finished building psp102.va in 1.46s
Finished building juncap200.va in 0.35s
Finished building psp103.va in 2.62s
Finished building psp103_nqs.va in 5.85s
Finished building psp104.va in 2.76s
Finished building psphv.va in 4.65s
Finished building r2_cmc.va in 0.07s
Finished building r2_et_cmc.va in 0.08s
Finished building r3_cmc.va in 0.24s
Finished building vbic_4T_et_cf.va in 0.23s
Finished building vbic_1p3.va in 0.33s
Finished building vbic_1p3.va in 0.35s
done
```

I had to change openvaf to openvaf-r in the script.

tomhajjar, 2024-12-01:
Where is the "new" openvaf-r for Windows?

As stated earlier, I need to test on a different Windows PC that doesn't have MS tools loaded.

ra3xdh, 2024-12-02:
Reopened. Everything seems to work with `openvaf-r` The library of the Verilog-A defined devices may be implemented in the future. The #980 will be a part of this feature.

ra3xdh, 2025-05-18:
I have recently introduced the library of SPICE subcircuits `share/qucs_s/spicelibrary` The advantage is it doesn't require the OpenVAF compiler installed. The attaching the Verilog-A models to a device may be implemented as a part of #1047 Closing this separate issue.

closed 2025-05-18 by ra3xdh