ra3xdh/qucs_s #344: Pressing "Simulate" opens a blank design window

yurivict, 2023-10-29:
Please see the video screenshot: https://people.freebsd.org/~yuri/qucs-s-empty-simulation-window-2023-10-29.mp4

Versions: 2.0.0, 2.1.0
FreeBSD 13.2

ra3xdh, 2023-10-29:
The simulation dialog was redesigned in the version v2.1.0 The Simulation button was removed and it doesn't allow to launch the simulaiton twice. Are sure you are using the v2.1.0? You should see the following simulation window with three buttons:
[image]

ra3xdh, 2023-10-29:
See also #309

yurivict, 2023-10-29:
The original video screenshot was made with the version 2.0.0
2.1.0 has different buttons, but it still opens the empty tab.

ra3xdh, 2023-10-29:
Yes, I understood now. You meant DPL tab. It is not a bug. You have to uncheck the "open data display" option in the *File->Document settings* dialog. In is not active by default for new schematic since the v2.1.0 But you have to change this option manually for schematics from previous versions.

yurivict, 2023-10-29:
Unchecking this option works.

But this is a very confusing behavior that by default it opens a blank tab.

closed 2023-10-29 by yurivict

ra3xdh, 2023-10-29:
The purpose of the blank tab is to place diagrams on it. This behavior was default before v2.1.0 See #309. Try to create a new schematic in v2.1.0 and you will see this option unchecked in the document properties.