ra3xdh/qucs_s #345: Math equation/functions syntax used for making Libraries

tomhajjar, 2023-10-30:
I'm having "syntax" issues making library models containing Equations and mathematical functions used to generate circuit parameters based on User inputs. 

It seems Qucs-S wants "Qucs syntax" for making the Library models yet ngspice requires Nutmeg syntax for analysis. In the past we discussed that Qucs and ngspice coincidently share some of the same function names allowing some legacy Qucs Equations in older Qucs schematics to work in Qucs-S instead of converting to Nutmeg Equations. I need to know what syntax to use for Library versus Analysis.

For example, Ideal_LPF1_model.sch has a Qucs equation that works when the model is used in Ideal_LPF1_lib_test.sch. When I try to test the schematic of the library device in Ideal_LPF1_model_test.sch, the equation fails.

In the case of library device Ideal_LPF2_model.sch used in Ideal_LPF2_lib_test.sch, none of the formulas I tried worked. The schematic Ideal_LPF2_model_test.sch doesn't work as well.

[Ideal_Library_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/13199708/Ideal_Library_ngspice_prj.zip)

ra3xdh, 2023-10-30:
Nutmeg equations are evaluated only after the simulation. SPICE has such division. Nutmeg equation has no effect inside subcircuits and libraries, because libraries and subcircuits evaluated before the simulation. You have to use `.PARAM` section instead if you need access to full set of SPICE syntax. This device is located in *SPICE specific section* group.

tomhajjar, 2023-10-30:
This is resolved.

ra3xdh, 2023-11-01:
Closing as resolved.

closed 2023-11-01 by ra3xdh