ra3xdh/qucs_s #349: Modify SPICE RCLDQ devices to provide support for microelectronics PDKs

ra3xdh, 2023-11-01:
I have recently added unified MOS devices that could be mapped to X letter. See #128 and #242. Other primitive devices also need to be modified to ensure compatibility with PDKs. It's need to add possibility to define pins number and letter. The following improvements should be done:

- [ ] Add terminal number and letter (C,X) property for capacitor  (needed 3-terminal capacitor)
- [ ] Add terminal number and letter (R,X) property for resistor (needed 3-terminal resistor)
- [ ] Define letter (D,X) and pins count for diode (needed 3-terminal diode)
- [ ] Add terminal number and letter (Q,X,N) property for BJT

Radvall, 2024-01-04:
I think more convenient way to use PDK in QUCS-S is to create a user library, rather than specifying the model name and model parameters for each element. I have no objection to such an approach, for a dozen elements it's OK. For a larger number, it seems to me that the library is better.

As an example of user library I am attaching a QUCS library for IHP-Open-PDK https://github.com/IHP-GmbH/IHP-Open-PDK. This library includes only resistors. The simulation are shown bellow:

[image]

[IHP_Open_PDK_QUCS_S_user_test_lib.tar.gz](https://github.com/ra3xdh/qucs_s/files/13834499/IHP_Open_PDK_QUCS_S_user_test_lib.tar.gz)

ra3xdh, 2024-01-05:
> I think more convenient way to use PDK in QUCS-S is to create a user library

Yes, I agreed with this. Furthermore we need to implement something like https://github.com/Qucs/qucs/issues/659 This will allow to put devices on the left panel without modifying C++ code.

closed 2024-01-05 by ra3xdh

ra3xdh, 2024-01-05:
Closed via #431. I put new 3-pin devices into the new group called "Microelectronics"

commit f40c93a3 (2024-01-05): Merge pull request #431 from Radvall/modify_spice_devices

Modify SPICE RCD devices; fixes #349