ra3xdh/qucs_s #356: Usage of Verilog-A modules in subckt's

dwarning, 2023-11-11:
I try to reproduce and understand the approach of qucs-s and Verolog-A models in the Verilog-A_EKV_2v6_prj project.
All the examples with flat netlists are working well e.g. the inverter in dc or tran simulation. But start the simulation with the hierarchical netlist (main circuit call's inverter subckt) follow in failure 
Error on line:
  n.xsub1.nekv26pmos1 out in _net0 _net0 xsub1:mod_ekv26pmos_ekv26pmos1 
could not find a valid modelname
    Simulation interrupted due to error!
The reason is that the pre_osdi load commands are not inserted into the netlist. This is done for all flat netlists. So the question is by which operation the pre_osdi lines are inserted? Is it "Project/Load Verilog-A module..."? I made this but w/o success.

ra3xdh, 2023-11-11:
The procedure of Verilog-A model usage could  be found here: https://mos-ak.org/venice_2014/publications/T_4_Brinson_MOS-AK_Venice_2014.pdf The OSDI files are produced with *Project->Build Verilog-A module*. The version 2.0.0 has the bug related to Verilog-A devices simulation #321 It is fixed in v2.1.0 I didn't check the case when Verilog-A devices are used inside subcircuits. But it should be no difference.  Sorry, we have no better documentation for Verilog-A related features.

ra3xdh, 2023-11-11:
You can also specify the `pre_osdi` directive manually using the spiceinit #26 if nothing helps. Sorry, this feature is not well documented too.

dwarning, 2023-11-11:
My version is 2.1.0 build from github 1 week ago.
Thanks for the workaround.
It is a bit tedious because of the missing working directory concept we have to specify full path 
osdi /home/dietmar/Dokumente/qucs-s/Verilog-A_EKV_2v6_prj/EKV26nMOS.osdi
osdi /home/dietmar/Dokumente/qucs-s/Verilog-A_EKV_2v6_prj/EKV26pMOS.osdi
in .spiceint. See #307.
My question remains: Which command fills in the pre_osdi commands in flat circuits and why not in hierarchical circuits.

ra3xdh, 2023-11-12:
There is no explicit device/command in Qucs-S to specify the OSDI file load. The netlist builder scans the schematic and adds the `pre_osdi` if the Verilog-A devices found. But it seems the netlister doesn't scan subcircuits recursively. It is a bug. I will provide a fix soon. This is not related to #307

ra3xdh, 2023-11-12:
It's need to move OSDI files search to `Ngspice::collectSpiceInit()` in order to fix this issue.

ra3xdh, 2023-11-12:
I have added a fix for this issue. Check the latest commit on the `current` branch. Now the Qucs-S always loads all OSDI files found in the project directory. The schematic must be opened from the project. The simulation of the Verilog-A devices outside the project is supported only using the .spiceinit features.

dwarning, 2023-11-12:
I removed the former inserted spiceint cmd's and made a new load va-module and it works perfect.
Thank you.

closed 2023-11-12 by dwarning