ra3xdh/qucs_s #362: TransformerPS1S2 Issue

tomhajjar, 2023-11-14:
I'm having issues trying to get a 6-phase rectifier circuit to work. I sent the netlist over to the ngspice guys and they found a possible error with .SUBCKT Transformers_TransformerPS1S2. The netlist line "R3 nSPlus nSplus  {RS1}" is wrong. There may be other errors in the .SUBCKT.

https://sourceforge.net/p/ngspice/discussion/133842/thread/c4ecc7c227/

[image]

[6_Phase_Rectifier.zip](https://github.com/ra3xdh/qucs_s/files/13351487/6_Phase_Rectifier.zip)

tomhajjar, 2023-11-14:
Holger recommendation is as follow. I cannot find "chgtol" in the Qucs settings.

You may add this line to the netlist. 
.options abstol=10u chgtol=1e-10 method=gear
 Because you have high voltages, chgtol and abstol reduce the precision requirements, especially the gear method here yields a transient solution. It looks strange though, and you should simulate up to 1 s to see the output signal decay.

ra3xdh, 2023-11-14:
Yes, the options helps. You may use the .OPTIONS device. 
[image]

tomhajjar, 2023-11-14:
Output is wrong. Supposed to look like the following.
[image]

[image]

tomhajjar, 2023-11-14:
Holger said to use Delay instead of Phase for the AC source. This and the changes to Transient Parameters seems to solve the issues. 

Can the "Qucs" AC source have VO (offset) and TD (delay) added as a parameter? The "Spice" AC source usage is a bit confusing. 

Attached is the "good and bad" schematics. As shown below, when using Delay in the AC source, the 120 and 240 Degree transformer outputs had a +/-200 volt DC offset. Unsure how this is possible. The offset occurs even if the diodes are removed.

Waiting for Holger to verify if this is a bug. 

[image]

[6_Phase_Rectifier.zip](https://github.com/ra3xdh/qucs_s/files/13356392/6_Phase_Rectifier.zip)

ra3xdh, 2023-11-15:
> Can the new Spice "ac Voltage Source" have VO (offset), VA (amplitude), FREQ (frequency), TD (delay), THETA (damping) and PHASE (phase) added as the default entry parameters? 

Yes, you may use the generic SPICE source for this purpose: 
[image]

> Dietmar pointed out that the "gnd" entry  is unused. Is this a mistake?

No, this is done intentionally for Qucs compatibility. This node never caused problems before.

tomhajjar, 2023-11-15:
The "ac Voltage Source" at the bottom of the Sources menu is a Spice only component so Qucs compatibility isn't an issue. It has a single default entry of "Vac" which is wrong. ngspice thinks it is a DC source. Why was this device created when "V Source" is available.

Can you alter "ac Voltage Source" by adding all the parameters to the Menu so the User can change them? 

[image]

[image]

[image]

[image]

ra3xdh, 2023-11-15:
Yes,  I have just added the TD and VO parameters for AC source. This fix will be available since the new release.

[image]

tomhajjar, 2023-11-15:
The "ac Voltage Source" at the bottom of the Sources menu will still be wrong if not changed or deleted. Why was it created?

[image]

ra3xdh, 2023-11-15:
> Why was it created?
This device was added by @MikeBrinson long time ago. I cannot say what was the purpose for this device. I am also finding two type of source confusing. I may delete it in the next release. The *V source* device would be sufficient if one need full SPICE definition.

ra3xdh, 2023-11-27:
I have deprecated the red AC source. The remaining issues with the transformer library will be fixed in #367 Closing this as completed.

closed 2023-11-27 by ra3xdh