ra3xdh/qucs_s #363: Changes to improve usability

medwatt, 2023-11-14:
I mostly use LTSpice for simple simulations and Cadence tools for integrated circuit designs. Cadence is very powerful but requires lots of clicking (it's also not free). LTSpice is free, simple to use but can be irritating when you want to run multiple tests. Besides it doesn't have support for verilog-a, which for someone that likes experimenting, is a big deal. I wanted a simple to use simulator that allows running multiple tests easily. This is how I came across Qucs. I've been enjoying it but I feel it lacks a few things that could improve its usability greatly.

I know that @ra3xdh has already done a lot. So please consider these as just suggestions:

1. My biggest gripe is the lack of a proper waveform viewer, specifically the lack of ability to zoom along an axis, and to have vertical and horizontal cursors. Having these would make experimentation much better.

2. The lack of customizable keyboard shortcuts really hurts usability. For example, in LTSpice, if I want to create an R C circuit, I press R, C, and V and within 10 seconds I have something ready. In Qucs, a lot of clicks are needed before a schematic is ready. A keyboard shortcut manager would really be nice.

[image]

3. The wire tool is very finicky. Sometimes I have to zoom in a lot to prevent the wire from bending. In LTSpice, the wire tool is rock solid because it only allows you to move in one direction at a time. A second issue with the wire tool is that if I run a wire through from one pin to the other pin through the component, the tool should know that I don't want to short the two pins together. This is the how the wire tool works in LTSpice and makes it easy to wire up a schematic.

4. When you zoom with the mouse, in almost every program that I've used, it is the region around the mouse pointer that gets zoomed. In Qucs, it is the top left corner.

5. There are too many voltage symbols for voltage sources. I want to do a DC simulation, I need a DC voltage source, a transient simulation, I need an AC source, an AC simulation, I need an AC source. LTSpice has only 1 voltage source that allows you to do all these things. See the figure below. You don't need to remember any SPICE code. I know that qucs has a V Source that allows you to input SPICE to define your waveform exactly as you want, but who cam remember SPICE code for a pulse waveform without looking it up? Does the delay come before the rise and fall time, do I need to put a zero here, can I leave it blank, etc. It is a very slow process.

[image]
 
6. Making simple plots. A lot of times, when you run an ac simulation, you want to plot the signal at a node with respect to the AC source. In LTSpice, you simply click on the code and that's it. In qucs, you have to write an equation, such as `db(V(out)/V(in))` before you can even see the response. Furthermore, there's no easy way to do FFT.

7. Saving transistor parameters by default. Whenever I do designs with MOSFETs, I have a look at parameters such the gm, gds, etc of the MOSFET. This is done by default in LTSpice, where all you need to do after a simulation is press `Ctrl+L` to see a log of parameters of all the transistors.

ra3xdh, 2023-11-15:
Thanks for the feedback! Here is my explanation on proposed improvements.

> My biggest gripe is the lack of a proper waveform viewer

I have explained my position here: #352 In principle I am agreed that Qucs-S would benefit from adding a waveform viewer. Also there were plans to redesign the plotting system using Qwt toolkit. But it's need to assign a full time developer for this task. It is not a weekend project. Currently Qucs-S has no full time developers. So, don't expect the waveform viewer to be implemented in the next years. 

> The lack of customizable keyboard shortcuts really hurts usability.

There exists unfinished shortcut manager for Qucs https://github.com/Qucs/qucs/pull/558 It's need to evaluate its status. 

> The wire tool is very finicky. 

This may require redesign of the custom widget representing schematic. Again full time developer needed. 

> When you zoom with the mouse, In Qucs, it is the top left corner.

I cannot reproduce this. The zooming area is centered around the mouse pointer. We have added some new zooming modes in the recent update.

> There are too many voltage symbols for voltage sources.

I don't agree with this. The different sources are more convenient in my opinion. You may use unified SPICE source if required. Regarding the TD parameter for Sine source, I can add it in the next update.

> In qucs, you have to write an equation, such as db(V(out)/V(in))

The equations approach was proposed by the previous developers team. I am making some steps in order to get rid of equations like dB. I have added a possibility to define dB scale for Y-axis:

[image]

> Furthermore, there's no easy way to do FFT.

Please explain what do you mean. I have added a spectrum simulation already in 0.0.24 See #88 What a more easier way to define FFT do you mean? I can add an automatic generation of the dB spectrum for existing simulation. 

>  In LTSpice, you simply click on the code and that's it. 

Yes, this would be a great improvement. But see my comment that there are no full time developers in the project now. 

> Saving transistor parameters by default.

I will look what could be done here without a need of the deep redesign.

medwatt, 2023-11-15:
> I cannot reproduce this. The zooming area is centered around the mouse pointer. We have added some new zooming modes in the recent update.

Thanks for your reply. Here's how zooming works with the mouse on my end. All I'm doing is holding down control and using my mouse wheel.

[image]

Here's how I expect it to work. This is from LTSpice.

[image]

> I don't agree with this. The different sources are more convenient in my opinion. You may use unified SPICE source if required. Regarding the TD parameter for Sine source, I can add it in the next update.

In LTSpice, you can change a source without having to change the symbol. How is this less convenient?

[image]

Presently if I want to apply a signal to a node, I need to use a DC Voltage and an AC voltage because the AC voltage does not have an offset parameter. Also, the name "AC voltage" is very confusing because AC voltage means, in most of the software I've used, the voltage you apply during AC analysis. Incredibly, there are two AC Voltage sources to make it even more confusing. Why not call the one used in transient as "Sine Source" and the one used in AC as "AC source". The unified SPICE source is nice to have only if you can have a GUI interface that does not require the user to remember the SPICE syntax.

[image]

> Please explain what do you mean. I have added a spectrum simulation already in 0.0.24 See https://github.com/ra3xdh/qucs_s/issues/88 What a more easier way to define FFT do you mean? I can add an automatic generation of the dB spectrum for existing simulation.

Last time I tried to do FFT, I search this forum and the way that was recommended was to use raw ngspice code where I inform ngspice to do FFT, which I found a bit inconvenient.

zergud, 2023-11-15:
1. whell zoom to mouse pointer is issue - yes
2. diagram zooming and markers is two different tasks. zooming planned to next release. markers TBD
3. ability to create a RC circuit in 10 seconds is not killer feature for me. but we will think about how to improve the process of creating circuits.

medwatt, 2023-11-15:
>     3. ability to create a RC circuit in 10 seconds is not killer feature for me. but we will think about how to improve the process of creating circuits.

The point I was making is that it is much faster to get components in LTSpice.

[image]

In qucs-s, you have to move your mouse from the schematic area to the components tab, down the components search bar, search, then move your mouse all the way to select the component. It's a very slow process if you have to repeat it many times for other components.

[image]

zergud, 2023-11-15:
I suggest next shotcuts combination
C (open components panel and focus on search textbox) + "cap" +  ENTER
L (open library panel and focus on search textbox) + "nmos4" +  ENTER

another options for quick reuse elements is "favorites" or (and) "last used" component lists

ra3xdh, 2023-11-15:
> Here's how zooming works with the mouse on my end. 

Yes, I understood now. The zooming doesn't follow the mouse pointer on sequential zoom. Don't expect a quick fix here, because the legacy code may be responsible for this.

> In LTSpice, you can change a source without having to change the symbol.

The redesign of the existing sources and designing new ones is not considered. But I can add VO and TD parameters for the sine source. The sources were designed to provide compatibility across Qucsator/SPICE and this compatibility should be preserved. Qucsator still have many unique features for RF simulation and we are intending to still maintain it. 

> there are two AC Voltage sources to make it even more confusing

The red devices should be used only if you need additional SPICE parameters.  The blue unified devices are sufficient for the most of cases. 

> The point I was making is that it is much faster to get components in LTSpice.

The starting point should be to evaluate the existing shortcut manager status https://github.com/Qucs/qucs/pull/558 I would put this task as a low priority. 

> Last time I tried to do FFT, I search this forum and the way that was recommended was to use raw ngspice code

This approach is totally outdated. You should use a new simulation type (*Spectrum analysis*) for FFT. Sorry, but the documentation update was delayed. Writing the documentation requires much more time effort than writing the code.

ra3xdh, 2023-11-15:
I have added the VO and TD parameters for sine source by the recent commit on the `current` branch. The new parameters will be available since the next release.

medwatt, 2023-11-15:
@ra3xdh, just merged the changes. I tried it and got this `V4 in 0 DC 0.354 SIN(0.354 0.5M 1K 0 0 0) AC 0.5M ACPHASE 0`. Why are you setting the ac signal to 0.5m ? The AC signal is used for AC simulation. It is usually set to 1 so that you don't have to scale the measured ac values.

ra3xdh, 2023-11-16:
> Why are you setting the ac voltage the same as the transient voltage?

Everything is correct. It is done to provide compatibility across Qucsator/SPICE. Qucs application never used separate setting for AC and TRAN voltage. Look at legacy schematics. This setting will not be reconsidered.

ra3xdh, 2023-11-16:
Here is the outcome of the discussion:

* The adding waveform viewer or waveform probe may be useful, but not possible in the near future. This task requires full time developer to be assigned and cannot be performed in the current time
* The adding LTspice style voltage sources is not considered. The backward compatibility should be  priority. I have added some SPICE parameters to AC source.
* The FFT analysis was added long time ago #88 Need to update the documentation
* Shortcut manager may be added as a low priority task. See #368 You may track the status here
* The solution for `save` support provided in #357  See my examples in comment.

closed 2023-11-16 by ra3xdh

medwatt, 2023-11-16:
@ra3xdh, if the transient voltage is the same as the ac voltage, does that mean that every time I do ac analysis and I want to know the voltage at some node `out`, I need to create an equation `db(V(out) / V(in)`, where `in` is the node where the voltage source is connected?