ra3xdh/qucs_s #365: Symbol generation from Verilog-A model

dwarning, 2023-11-15:
I used the short description from Mike Brinson's Venice presentation PDF as a tutorial to generate a symbol from Verilog-A model ekv26.va
I can compile the model with openVAF and generated the Verilog-A user device. But if I want use it in the schematic by click and go to schematic not the transistor model appears. Only the template box is there.
What is wrong with the approach? Project file is attached.
[image]
[Verilog-A_ekv26_prj.zip](https://github.com/ra3xdh/qucs_s/files/13369326/Verilog-A_ekv26_prj.zip)

ra3xdh, 2023-11-16:
It seems you forgot to edit and save the device symbol. After pressing F9 on the Verilog-A source the symbol editor opens. I have seen only square symbol in your project. After I edited and saved the *.sym file everything works as expected. 
[image]
[image]

dwarning, 2023-11-16:
I made F9 and the ekv26.sym was opened but as a box. I mirrored few things and made save.
But from where I should get the transistor symbol as you shown in your last picture? You really paint it with the primitives paintings on left side? Can't believe. Is there a documentation?
Sorry for my very basic question.

ra3xdh, 2023-11-16:
Yes, you must draw the symbol from primitives. Automatic symbol generation is not possible. But we provide symbol templates with the application. Open the  `share/qucs-s/examples/symbols/schematic_symbols.sch` document and copy-paste the symbol. 
[image]

dwarning, 2023-11-16:
Yes - now I got it.
Thanks,
I will close

closed 2023-11-16 by dwarning