ra3xdh/qucs_s #367: XSPICEPS1S2 transformer issues

tomhajjar, 2023-11-16:
I found more problems with XSPICEPS1S2 in Transformer.lib. I made three circuits comparing XSPICEPS1S2 and TransformerPS1S2 in Transformer.lib to Three Mutual Inductors in Components. Analysis of TransformerPS1S2 and Three Mutual Inductors generally agree.

XSPICEPS1S2 has two issues:

1. Rp, Rs1 and Rs2 generate warnings. Is there node naming issues like in TransformerPS1S2?
2. The "Turns Ratio" is half what it should be. You can see the output voltages and currents in the plots are half compared to transformers TransformerPS1S2 and Three Mutual Inductors.

[image]
[image]
 
[Transformer_Lib_Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/13371508/Transformer_Lib_Test_prj.zip)

ra3xdh, 2023-11-16:
> Rp, Rs1 and Rs2 generate warnings. Is there node naming issues like in TransformerPS1S2?

No, the variable should be wrapped in braces like `{Rs1}` Otherwise Ngspice may interpret it as the model.

tomhajjar, 2023-11-16:
Holger corrected the XSPICEPS1S2 subcircuit. AFAIK, the info in the online help is wrong.

I found another issue, Cores.lib uses the same data for every core so only one core is correct....

ra3xdh, 2023-11-16:
> Cores.lib uses the same data for every core so only one core is correct..

Yes, I have also noticed that the Cores.lib uses duplicate data. Unfortunately, I have no source data for this library and cannot provide the fix. The library was developed by @MikeBrinson around 2015. I cannot say where did he take the cores data.

ra3xdh, 2023-11-16:
> AFAIK, the info in the online help is wrong.

The help on readthedocs.io should be deprecated and replaced by the new tutorial. It contains a lot of outdated information. I have started to collect the new documentation here: https://github.com/ra3xdh/qucs_s_docs But this a slow process. Writing documentation requires much more time effort than writing a code.

tomhajjar, 2023-11-16:
Here is the updated Transformers.lib. All identified errors have been fixed.

I left the "Cores" data in Transformers.lib and Cores.lib as it was. I could not find any info on the web. Looks like manually pulling data from B/H plots would be required. Some of the cores are nothing a user would ever make a transformer.

I requested info from Mike Brinson...

[Transformers.zip](https://github.com/ra3xdh/qucs_s/files/13383341/Transformers.zip)

ra3xdh, 2024-01-09:
I have uploaded a new `Transformer.lib` library. Thanks for the contribution! Closing this.

closed 2024-01-09 by ra3xdh