ra3xdh/qucs_s #369: Option to remove redundant information in schematics

csrabak, 2023-11-16:
In present implementation the inclusion of lumped componentes in the schematic with the value of the components adds a letter and an equal sign, which is redundant since the symbol already is showing the component type, and if one decides "not to show" in properties pane, it removes the whole information altogether, so, say for a resistor, you have or **R1 R = 30 Ohm** or **R1** without the value.

I think this R =, C =, L = are redundant, and more useful is the maintaining of the auxiliary information on the schematic, like the Q of coils, other model parameters, like series resistance, etc.

On the other hand, when inserting active devices, like BJT, the component label ends up with the whole (JEDEC, or EU) designation added with and underscore, it would be more natural to have Q1, Q2, etc, and the part number placed likewise the component value.

csrabak, 2023-11-17:
@ra3xdh :

Let me test my understanding of your assertion: do you mean that when a component, say a resistor, has a value near it, it is not redundant to have an expression saying "R = " ?

More explicitly: just have the component symbol and the value of the main parameter is not enough?

My wondering is not without reasoning, as you bolded the word "all" with relation to backward schematics (which I surmise refers to qucs ones) as be it other schematic capture software as well as present graphic representation technique not to have this appendage.

ra3xdh, 2023-11-17:
In Qucs devices has no values like in KiCAD. Every device has a set of properties instead. The "R=" is a property, but not a value.  This is a basic concept of the application since 2003 and it is not subject of a discussion and change. The changing of this concept will break many things.

zergud, 2023-11-17:
actually this is a bit complex issue, related to #92 "Add automatic enumeration of the devices on schematic (like KiCAD etc., low priority)"
steps to implement (abstractly)
1. add RefDes (Reference Designator) property for components (Q for transistor, D for diode etc.)
implement enumeration of the devices by this property
(current "name" property remains untouched)
2. refactor property window and property itself
move currnent "name" to property
add option like "show name" for property
3. maintain backward compability 

don’t see anything impossible, I suggest we do it later (due lack of time)

ra3xdh, 2023-11-17:
> actually this is a bit complex issue, related to https://github.com/ra3xdh/qucs_s/issues/92

If the implementation of the request requires a massive refactoring I would simply remove it from the task list. Automatic device enumeration doesn't directly affect the circuit simulation itself. Currently the component RefDes is represented by `Component::Name` Moving it to the properties may  require to rewrite many thing. The generation of the spice netlist entry depends on `Component::Name` field. Also don't forget qucsator netlist that are mostly generated by unified method `Component::netlis()`. The task may become much more complex than it seems. So, I am inclined to avoid massive refactoring if possible.

csrabak, 2023-11-17:
I will make this a question as I didn't delve in the code, and accepting all the arguments at face value and valid:

Can we maintain the architecture of the properties (like @ra3xdh shows in the example of "R = ") but have the method that prints in the schematic canvas format it to show only the "contents" (I'm trying to avoid an amphibology if the word _value_ is used)?

In fact I surmise the code is almost there, as Vadim mentions the export of Spice netlist and in the exported list the values are just put in the correct place in each row of the netlist.  Couldn't it be lifted or factored in such a way it can make the funcionality to put in the canvas?

Again I thank everybody who engaged in this conversation.

ra3xdh, 2023-11-18:
> Can we maintain the architecture of the properties

No, i don't consider this option. It is not as simple as it seems. The introduction of the "show/hide" option for the property name will require its saving in the schematic file. This will require the modification of the schematic XML format that may cause side effect. Also Qucsator doesn't have the concept of the device value. The compatibility across the simulators should be preserved and backward compatibility is the priority. The current device design exists almost 20 years and is not a subject of change. Accept this.

On the other hand, other circuit simulation software like Cadence Virtuoso uses the similar concept of the device properties. Probably original design of Qucsa devices was inspired by Cadence. 

[image]

zergud, 2023-11-18:
In theory:
For now visibility saved as 0/1 in xml
Ee can add option 2 - show only value and maintain backward compatibility 
Yes/no already combobox in UI

ra3xdh, 2023-11-18:
> Ee can add option 2 - show only value

Yes, this may be a solution. But I don't recommend to touch the basic C++ classes structure because of cosmetic improvement. We have a more important tasks that directly affect the simulation at the current time. Old Qucs version had more than 1000000 downloads and nobody complained about show/hide parameter name more 20 years.

csrabak, 2023-11-18:
I think the "cosmetic" is in the eyes of the beholder...

My coleagues attempted to capture this schematic in  Qucs-s [https://www.diyaudio.com/community/threads/100w-ultimate-fidelity-amplifier.164093/](url), it gets busy very fast, now consider:

[image]

Notice that the value "Rser = .1" adds information as the component is a capacitor, whereas the value of it is written without additional stuff and device designation C6 plus symbol is plainly enough to convey information.

On the other hand I've not power of attorney for the other one million people who downloaded the old qucs SW, but I think the argument is not robust as it seems, because we cannot insure which percentage of the downloaded packages have been put to use, and more importantly are still being in use today.

Pertaining the comment on Cadence Virtuoso, which I've not access: does the "r1: 1K" shows this way in the schematic canvas or is this the rendition of the pane for editing the properties of symbol alone?