ra3xdh/qucs_s #372: Experiences in implementing VA modules

dwarning, 2023-11-18:
The attached project file shows the modified IGBT model (Hefner et. al.) in co-work of qucs-s/ngspice.

Main issues are:

- Symbol generation reading out parameter definition (see also #366). All `parameter` strings in comments have to change, only pure `parameter real name=value` are accepted. Trial in adding parameter descriptions (desc) follows in symbol not found or crash of the program.
- openvaf has the capability of `-DMACRO` cmdline parameter. Please add this option for calling openvaf.
- in the tran parameter specification are options from other simulators left over. Please adapt it to ngspice specific tran options. Also `uic` have to be separated from `start/maxstep` spec.
- the coil is of-grid - have to change grid scale from 10 to 2.

[image]

[image]

[Verilog-A_IGBT_prj.zip](https://github.com/ra3xdh/qucs_s/files/13399891/Verilog-A_IGBT_prj.zip)

ra3xdh, 2023-11-18:
> in the tran parameter specification are options from other simulators left over

Please explain. The TRAN simulation is unified simulation. The incompatible options cannot be removed. We can only mark them as Qucsator-only. 

> Also uic have to be separated from start/maxstep spec.

Please explain. You should set `InitialDC=yes/no` instead of UIC. 

> the coil is of-grid - have to change grid scale from 10 to 2.

Try right mouse button menu -> Align on grid or Ctrl+U shortcut. It is a known problem. The device may fall out of grid. We are working on solution #92  The fix may be not as simple as it seems.

dwarning, 2023-11-18:
OK - marking them as Qucsator only would be sufficient.

If I set InitialDC = no and set max time step I got:
tran 1e-08 0.0001 0 20n uic
If I set InitialDC = no and set no max time step I got:
tran 1e-08 0.0001 0  uic
If I set InitialDC = yes and set max time step I got:
tran 1e-08 0.0001 0 20n

So you see in my version 2.1.0 build from git the InitialDC is wrong.

What about the parameter parse problem? #366

ra3xdh, 2023-11-20:
>  build from git the InitialDC is wrong.

The InitialDC is not wrong. It uses the following logic. Everything according the section 15.3.10 of the Ngspice manual.

*  `MaxStep = 0` and `InitialDC=yes` outputs `tran tstep tstop tstart`
*  `MaxStep > 0` and `InitialDC=yes` outputs `tran tstep tstop tstart tmax` where `tmax=MaxStep`
*  `MaxStep = 0` and `InitialDC=no` outputs `tran tstep tstop tstart UIC` 
* `MaxStep > 0` and `InitialDC=no` outputs `tran tstep tstop tstart tmax UIC` where `tmax=MaxStep`

What logic for setting tmax and UIC should be applied?

> What about the parameter parse problem? https://github.com/ra3xdh/qucs_s/issues/366

This fix will require some time. Please wait.

dwarning, 2023-11-20:
I thought InitialDC means UIC (Use Initial Conditions) and I have to set on _**yes**_ to get it.

I think I found the file in question. In schematic_file.cpp line 270ff is the missing place for skipping unwanted strings in comments.
But the problem is more general: We have dozen of different forms in actual, modern CMC models, with and without macro using for instance and model parameter definition.
Maybe using a preprocessor would be easier.

ra3xdh, 2023-11-27:
Assume the questions are answered. The bug with VA-files parsing will be fixed in #366 Closing this ticket.

closed 2023-11-27 by ra3xdh