ra3xdh/qucs_s #373: Steps required to run OpenVAF in Qucs-S under Windows

tomhajjar, 2023-11-20:
I tried to get OpenVAF to work under Windows and failed. dwarning gave some help but I'm still having issues. 

openvaf.exe was placed in:
C:\QUCS-S 2.1.0\bin\openvaf.exe

I installed "Desktop development with C++" and got the same errors shown below as before the install.
I uninstalled "Desktop development with C++" and installed "Visual C++ redistributable package 64-bit" and still have same errors.

Why is some of the text in the Dock messages garbage? 

[image]
[image]

ra3xdh, 2023-11-20:
It is not garbadge output. OpenVAF uses control symbols to switch colors in the text terminal. I don't know what CLI option to apply for OpenVAF to supress the colorized output. @dwarning could you suggest? 

The "unexpected token" error is not related to Qucs-S. OpenVAF cannot compile the module by itself. Probably it contains some syntax that is not supported by OpenVAF or requires to apply specific option.

dwarning, 2023-11-20:
No idea how to switch of these output option.
Sorry, forgot to remove nigbt_org.va. It's the old version for admsXml and Qucs <= 0.19.
Please consider only igbt3.va.

ra3xdh, 2023-11-20:
The "linker not found" error also is not related to Qucs-S. Probably windows version of OpenVAF requires an external linker. Could you obtain the OSDI file if you run `openvaf.exe igbt3.va` from the CMD.EXE?

tomhajjar, 2023-11-20:
I got it to work. Definitely not user friendly like qucs 0.0.19...

I followed the info in the link sent by dwarning. The comment by "filiphagan" worked. Out of the dozens of choices I picked 3. It took 30min for the files to download and install.  

https://stackoverflow.com/questions/55603111/unable-to-compile-rust-hello-world-on-windows-linker-link-exe-not-found

ra3xdh, 2023-11-21:
ADMS was written in C++, but OpenVAF is wriiten in Rust. Rust may have difficulties in deploying the application on different platforms. Please consider reporting Windows installation issues to OpenVAF developer https://github.com/pascalkuthe/OpenVAF/ Probably he will put a more detailed instructions how to install OpenVAF on Windows.

ra3xdh, 2023-11-21:
I have just seen https://github.com/pascalkuthe/OpenVAF/issues/100 The OpenVAF developer is informed.

ra3xdh, 2023-11-21:
Closing as resolved.

closed 2023-11-21 by ra3xdh