ra3xdh/qucs_s #374: BiasT to lumped components

dwarning, 2023-11-21:
Is there a reason why this component was removed from _lumped components_ for ngspice simulator?
For qucsator the component is available. It is a simple but practical circuit which can be build by a LC subckt.

ra3xdh, 2023-11-21:
I didn't prepare a SPICE netlist for this device yet. That's why it still be hidden. I will try to fix the BiasT by the next release.

tomhajjar, 2023-11-22:
I never liked the Bias T graphics so I redid them. The new one is the same size and has the same Port locations.

[image]

[BiasT.zip](https://github.com/ra3xdh/qucs_s/files/13433271/BiasT.zip)

ra3xdh, 2023-11-22:
I didn't update the device in the source code yet. Reopening.

ra3xdh, 2023-12-01:
I have added SPICE entry for BiasT. The updated device will be available since the next release. Closing this.