ra3xdh/qucs_s #377: Verilog symbol generation for single Verilog file containing both nMOS and pMOS

tomhajjar, 2023-11-22:
Issue #365 left out the steps for generating both nMOS and pMOS models and symbols when the Verilog file contains both. 

Old example in Qucs 0.0.19 used separate ekv26nMOS.va and ekv26pMOS.va files. What is the procedure for ekv26.va?

Attached is the old Qucs project minus the old va files which don't work under OpenVAF.

[Verilog-A_EKV_2v6_prj.zip](https://github.com/ra3xdh/qucs_s/files/13439991/Verilog-A_EKV_2v6_prj.zip)

ra3xdh, 2023-11-22:
There is no solution for this issue using the existing procedure for Verilog-A devices. I am planning to add new Verilog-A device that will act like *SPICE file* device and will not require the JSON files. But don't expect this feature to be implemented in the near future because of low demand and high time effort. 

You can use the following workaround. 

* Compile the VA file from CLI outside Qucs-S using OpenVAF
* Specify the  OSDI path using the spiceinit script.
* Put .MODEL on schematic
* Put the *SPICE generic* (Z) device referencing letter N and specify the proper model name for it

tomhajjar, 2023-11-25:
The file usually has a flag of 0/1 or 1/-1 that determines n or p. One could add a line that sets the flag and create two files.

ra3xdh, 2023-11-25:
It is not possible to attach two JSON `*.sym` files to one `*.va` file using the existing workflow. Wait until the Verilog-A devices will be rewritten without need to use JSON or use the workaround. I don't promise a quick fix for this.

tomhajjar, 2023-11-25:
One could create two files and add a line that sets the flag.

I don't use Verilog myself...

dwarning, 2023-11-26:
@ra3xdh : Don't know your concept. But at the moment it is a bit cumbersome that each instance get his own model fixed burned in the netlist. This prevent to include library files with .model cards for each device construction and type. This concept can sufficient for usual rf circuits but not for analog and mixed-signal circuits.
Another topic is the separation of instance and model parameter. The latest release of CMC VA models make more and more this distinction, whenever unfortunately some models allow double usage - model and device parameter. I think it should considered to take care for that in the future.

closed 2023-11-27 by ra3xdh