ra3xdh/qucs_s #378: Issue running EKV26 Verilog example

tomhajjar, 2023-11-22:
dwarning uploaded a Verilog example of the EKV26 in #365. I get strange errors as shown below.

The netlist "EKV_2v6_idvd_nmos.net" has "comments" in ekv26.va stuffed into ".MODEL mod_ekv26_ekv261..."

instance=parameter, process=related, voltage/substrate=effect, parameters=(long-channel), sat./channel=length, related=parameters...

[image]

[image]

[image]

[Verilog-A_EKV26_prj.zip](https://github.com/ra3xdh/qucs_s/files/13442857/Verilog-A_EKV26_prj.zip)

ra3xdh, 2023-11-22:
It is related to #366 Wait for fix.

dwarning, 2023-11-23:
See my comment under #366.

ra3xdh, 2024-01-05:
You may use the workflow with generic device or 4-pin MOSFET proposed in #411. Closing as won't fix. Old workflow with JSON files should be kept only as backward compatible.

closed 2024-01-05 by ra3xdh