ra3xdh/qucs_s #385: Temp sweep ngspice

dwarning, 2023-11-25:
In ngspice we can make nested sweep with temperature in the outer loop:

`dc v1 0 1 0.01 temp -50 150 50`

or a separate loop:

`temp -50 150 50`

Is there a simple way to make this under qucs-s for ngspice simulator?

ra3xdh, 2023-11-25:
Specify the TEMP as DC sweep parameter. The blue unified devices  require to set `Temp=TEMPER` explicitly. The red SPICE devices will work without this setting. Look at the attached example. 
[image]

dwarning, 2023-11-25:
What you mean with blue and red transistors?
It looks your approach expect to have the Temp model or instance parameter. But there are models which haven't and get the temperature over the circuit temperature. For larger circuits it would be uncomfortable to edit any instance.
I found a simple way with Nutmeg script.
Thank you.
[image]

ra3xdh, 2023-11-25:
> What you mean with blue and red transistors?

* Blue transistors are unified devices. It has blue color on schematic. This transistor si supported by both Qucsator and Ngspice. This is blue transistor. It has blue color.
[image]

* Red transistors are SPICE only and support extended SPICE parameters. It has red color on schematic. This is red transistor. It requires the .MODEL to be attached. This transistor doesn't require the TEMPER to specified explicitly and uses global temperature. The picture shows red transistor. It has red color. 
[image]

> But there are models which haven't and get the temperature over the circuit temperature.

I don't know if Verilog-A devices take the global temperature. 

> For larger circuits it would be uncomfortable to edit any instance.

The solution may be don't use the unified devices if you need temperature sweep. I cannot provide an unified setting tor `Temp` parameter for blue devices. Qucsator will not accept `TEMPER` and the simulation will show undefined variable error. But the number is accepted by both Qucsator/Ngspice. I can propose the following logic.

* If `Temp == Tnom` don't put the Temp in the parameter list. The global temperature will be uses.
* If `Temp != Tnom` put the Temp in the parameter list. The device will be computed at the temperature specified in `Temp`

dwarning, 2023-11-25:
Ok, now understood.
The hicum2 version 3.0.0 has no external temp parameter and get the circuit temperature in the va code.
I changed the symbol colour belong your guideline:
[image]

A bit off-topic: Can I prevent the retraces in the plots? In ngspice we have the retraceplot as a plot option.

zergud, 2023-11-25:
> A bit off-topic: Can I prevent the retraces in the plots? In ngspice we have the retraceplot as a plot option.
can you explain in detail? give an example
i am not very familiar with ngpice

ra3xdh, 2023-11-25:
> Can I prevent the retraces in the plots? In ngspice we have the retraceplot as a plot option

It is not bug, but rather a feature. Qucs plotting engine cannot understand that you are using sweep in a nested loop in `DC ... TEMP` statement. When you are using parameter sweep simulation type Qucs-S creates some supplementary files. These files are needed to tell the plotting engine that this is a stack of plots but no single plot. The script simulation type cannot give this information to the plotting engine.  I think this cannot be changed without redesign of the plotting engine. But there may be a workaround. Try to sweep the temperature in while loop.

dwarning, 2023-11-25:
Accepted - there are other things with higher priority.

ra3xdh, 2023-11-27:
Closing as resolved.

tomhajjar, 2023-11-28:
Every Verilog example done in Qucs used "Blue", HTML "000080".

Native Spice components in Qucs-S uses QT "darkRed" or HTML "800000".

[image]

closed 2023-12-28 by ra3xdh