ra3xdh/qucs_s #386: S-Parameter ngspice plot

dwarning, 2023-11-26:
I try to make a s-parameter simulation with one point at different operating points:
[image]

The problem is that can't plot the results vs Vbe. The result file bjt_spar.dat.ngspice has reasonable values. But plot is nothing. Plot over frequency in sweep works ell.

Can some help me please.

[bjt_spar.sch.txt](https://github.com/ra3xdh/qucs_s/files/13468002/bjt_spar.sch.txt)

tomhajjar, 2023-11-27:
This may be the issue...

Qucs-S has issues plotting single frequencies, CW. In #156 Vadim used a 3D plot for a temporary fix. 

Qucs/ngspice use different methods to calculate simulation frequencies from Start, Stop, Step and Number. Vadim can explain better than I.

tomhajjar, 2023-11-27:
I found an old QucsStudio schematic that looks like your schematic. Unsure where I got it from. 

[image]

ra3xdh, 2023-11-27:
@dwarning It is a known issue and explained in #156 and #161 The plotting engine cannot assemble a set of 1x1 plots in a smooth curve. Two solutions are possible:

* Set line style to Circles/Stars in the diagram properties. You will see the simulation points. Also you may use the table.
* Use the scripting simulation to assemble a vector from set of 1x1 plots. I have uploaded a CV-curve simulation example that uses this approach. See #156 for more info.

closed 2023-11-27 by ra3xdh