ra3xdh/qucs_s #389: OP annotation with ngspice

dwarning, 2023-11-29:
Is it possible to show the results of an OP analysis with ngspice?

If I make only dc analysis the was no op statement in netlist so in schematic only 0 stands in schematic:
[image]

Log file is complaining of missing model.
If I use the Nutmeg script block analysis is running and print all too but no annotation.

ra3xdh, 2023-11-29:
It is a known issue. This feature is not supported. SaveOP has no effect. See #124 I can suggest only to use F8 or 1 point DC sweep with table output. Zero output on the show circuit is not related to applied simulations. There is something wrong in the model itself. Look in the logs (press F5)

ra3xdh, 2023-11-29:
Closing as duplicate of #124

closed 2023-11-29 by ra3xdh