ra3xdh/qucs_s #395: Replace FreeHDL by GHDL

ra3xdh, 2023-12-09:
Qucs-S uses FreeHDL for VHDL simulation. The FreeHDL is deprecated and doesn't work on modern platforms anymore. It's need to replace it with GHDL.

zergud, 2023-12-10:
Qucs/qucs#830

ra3xdh, 2024-01-29:
It seems all VHDL models require a revision after switching to GHDL. I god a simulator freeze trying to simulate simple schematic containing one demultiplexer device. GHDL gives warning about infinite loop. The same schematic could be simulated with iVerilog without warning. Simple schematics containing logic gates work without problem using GHDL. FreeHDL is crashing on every simulation on my machine, so I cannot compare the result. 

Unfortunately, digital VHDL/Verilog design is not my field of expertise. The help in VHDL models revision is highly welcome.  

[image]
[image]

iwbnwif, 2024-01-30:
I don't pretend to know much about ghdl, but I think that each process must have some wait statement to give the simulator a chance to actually output something (unless it has a sensitivity list as is the case of the process which represents the demux block). 

In your example can easily work around this problem by appending "wait for 1 ns;" before the "end process;" in the Logic_0 process definition.

Having a quick look, it seems that none of the digital components that generate a pattern do this properly except for the 'digital source' which do to create the timing of their output toggle.

Simplistically, we could hard-code a "wait for 1 ns;" at the end of each relevant vhdlCode() function, e.g. in pad2bit.cpp, pad3bit.cpp, pad4bit.cpp, logic_0.cpp, and logic_1.cpp.

iwbnwif, 2024-01-30:
A quick test indicates the only problem remaining is with the R/S flip flop.
[image]

iwbnwif, 2024-01-30:
R/S flip-flop is working fine - it just didn't like both inputs being driven from the same source (e.g., undefined behavior at 1ns transitions)!

With a proper test, there is no problem:
[image]

closed 2024-02-10 by ra3xdh