ra3xdh/qucs_s #401: Update Matching Tool to Qucs 0.0.20 Version

tomhajjar, 2023-12-19:
Any plans to update the Qucs 0.0.19/Qucs-S Matching Tool to the version in Qucs 0.0.20? 

https://github.com/Qucs/qucs/pull/666 

The attached test schematics done by @andresmmera work with qucsator from Qucs 0.0.20. The planned qucsator_rf, should work as well.

[image]

[Matching_Tool_tests_prj.zip](https://github.com/ra3xdh/qucs_s/files/13717267/Matching_Tool_tests_prj.zip)

ra3xdh, 2023-12-19:
Yes, I will look how https://github.com/Qucs/qucs/pull/666 could be backported. This would be useful feature.

ra3xdh, 2024-03-25:
Matching tool backported by #662

closed 2024-03-25 by ra3xdh