ra3xdh/qucs_s #402: Regulators.lib and ngspice issues with LM317 spice models

tomhajjar, 2023-12-20:
I'm looking to redo Regulators.lib for multiple reasons as shown below. I will look into adding some popular regulators after the library gets fixed to work for both qucsator and ngspice. 

1. The PVR100AZ-xx "regulators" are obsolete, cost $6 and consist of a 100ma NPN transistor and a zener diode. No one would ever use them.

2. The two "LM317" devices LM317K-1 and LM317K-2 are almost identical as shown in #375. They both work with qucsator but LM317K-1 won't work for ngspice.
https://github.com/Qucs/qucs/issues/367
https://github.com/Qucs/qucs/pull/375

3. An old Qucs example LM317_prj.tar.gz works under qucsator but don't work for ngspice. Both Spice files lm317.cir and lm317_2.cir crash Qucs-S. 
https://qucs.sourceforge.net/examples.html

[image]

[image]

[LM317_prj.tar.gz](https://github.com/ra3xdh/qucs_s/files/13722852/LM317_prj.tar.gz)
[LM317_prj.zip](https://github.com/ra3xdh/qucs_s/files/13722855/LM317_prj.zip)

tomhajjar, 2023-12-21:
Tested more LM317 models and still no luck finding one that works for both qucsator and ngspice. Here is the latest test data...

[LM317_prj.zip](https://github.com/ra3xdh/qucs_s/files/13736673/LM317_prj.zip)

ra3xdh, 2023-12-21:
> The two "LM317" devices LM317K-1 and LM317K-2 are almost identical.  I cannot vouch for the accuracy of the models.

These models have unknown origin. I have plans to redesign this library and add true SPICE models for linear regulators. 

> Tested more LM317 models and still no luck finding one that works for both qucsator and ngspice

The most probably you will not find the model that will work well with both qucsator and ngspice. I suspect that the models from `Regulators.lib` were handwritten. The automatic SPICE to Qucsator converter is very poor and cannot convert complex models.

tomhajjar, 2023-12-21:
I'm having issues with an LM317 model that works with ngspice crashing Qucs-S. Attached is the ngspice files and Qucs-S files.

Model and circuit was provided by Holger in following link:

https://forum.kicad.info/t/lm317-regulator-model-import/33839/5

[LM317_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/13743559/LM317_ngspice_prj.zip)

ra3xdh, 2023-12-21:
> LM317 model that works with ngspice crashing Qucs-S

Yes, I confirm this bug. I will provide a fix soon. As a workaround try to remove `;OUT_1` node from the SUBCKT definition. The header should look like as the following: 

~~~
.SUBCKT LM317_TRANS IN ADJ OUT_0 
~~~ 
Also keep in mind that this model uses LTSpice extentions. It cannot be accepted for Qucs-S library unless the syntax will be rewritten.

ra3xdh, 2023-12-21:
> I will look into adding some popular regulators after the library gets fixed to work for both qucsator and ngspice.

I would deprecate this library and replace it with the new library containing true Spice model. Providing Qucsator compatibility is not needed here, because Qucsator has very poor performance for complex schematics.

tomhajjar, 2023-12-21:
These regs may be useful, LM7809, LM7909. They seem to work with ngspice. The symbols are useful.

https://forum.kicad.info/t/spice-models-lm78xx-lm79xx-series-edit-04-05-23/41489/5

[Voltage_Regulators_prj.zip](https://github.com/ra3xdh/qucs_s/files/13744504/Voltage_Regulators_prj.zip)

ra3xdh, 2023-12-22:
I have added TODO list for this issue.

tomhajjar, 2023-12-25:
Using the beta version of Qucs-S "24.1" I revisited testing the LM317 example from @holvo. The crash is gone but the simulations fail.

[LM317_TRANS_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/13766811/LM317_TRANS_ngspice_prj.zip)

ra3xdh, 2023-12-25:
>  The crash is gone but the simulations fail

Everything works on machine as expected. Try to fix the subcircuit name as shown on the screenshot. Otherwise it gives "Subcircuit redefined" warning. Also don't forget to set the LTspice compatibility option. The used model is LTspice model and will not work without setting `ngbehavior`.

[image]

tomhajjar, 2023-12-26:
Using DuSpice as the GUI for ngspice, the unmodified LM317_TRANS.LIB file works without any obvious warnings in the LM317_DC_sweep.log file. Where do the warnings displayed in Qucs-S come from? Since the files work in ngspice, shouldn't they work using Qucs as the front end?

[image]

[LM317.zip](https://github.com/ra3xdh/qucs_s/files/13772242/LM317.zip)

ra3xdh, 2023-12-26:
> Where do the warnings displayed in Qucs-S come from?

This question was discussed man times. Qucs-S GUI doesn't hide or add log warnings. It keeps the log as is. Run the Ngspice from `cmd.exe` or Linux terminal and you will see exactly the same log as shown in the Qucs-S GUI. I don't know what options may DuSpice apply so the logs are different.

ra3xdh, 2023-12-26:
> Using DuSpice as the GUI for ngspice, the unmodified LM317_TRANS.LIB file works without any obvious warnings

It is not Qucs-S bug. The model from `LM317_TRANS.cir` uses LTspice syntax. It contains the `IF` operator that is not supported by Ngspice unless the compatibility options is set. It will not work without setting LTspice compatibility mode. I suspect DuSpice may use the different path to `spiceinit` so the compatibility options are set implicitly. If the compatibility options is set everything works as expected.

[image]

tomhajjar, 2023-12-26:
I have a .spiceinit file with "set ngbehavior=ltpsa" in my "User" directory. DuSpice uses it.

tomhajjar, 2023-12-28:
Here is the first cut of the new Regulators.lib. More devices will be added in the future. The new libray is in Voltage_Regulators_lib_prj. 

1. One still can't make a new library called "Regulators.lib" even though Qucs places it in the user_lib folder. You cannot just rename the file after creation either since the term "Regulators" becomes the prefix of EVERY subcircuit in the library. One is forced to delete or rename Regulators.lib in "C:\QUCS-S 24.1.0\share\qucs-s\library" BEFORE making the library.

2. If during entering the device description you make a mistake and click "Previous", Qucs makes you re-enter EVERY device description TWICE. I ended up making a files with every device description so I could do a copy-paste. 

[Voltage_Regulators.zip](https://github.com/ra3xdh/qucs_s/files/13789314/Voltage_Regulators.zip)

ra3xdh, 2024-01-10:
Closing as completed via #413 and #439

closed 2024-01-10 by ra3xdh