ra3xdh/qucs_s #403: LDMOS simulation froze Windows

tomhajjar, 2023-12-22:
I'm trying to model an LDMOS power device and Windows froze doing a simple DC characteristic curve simulation. I was forced to do a hard reset.

On a side note I'm unsure if the ngspice Windows version supports Level 73 MOS devices. In the past I was told certain level MOS devices requires a custom version of ngspice. That never caused my PC to lock up or crash before however...

[image]

[MRF101_LDMOS_prj.zip](https://github.com/ra3xdh/qucs_s/files/13753917/MRF101_LDMOS_prj.zip)

dwarning, 2023-12-22:
The problem is the obsolete token "Nchan" before the brace.
.model MRF101 VDMOS Nchan(Rg=4 Rd=1 
It is a bug in ngspice that no notice was issued. Use the VDMOS as described in the manual.
Move the token after opening brace and it will work.

dwarning, 2023-12-22:
Another issue in ngspice is the very small ksubthres parameter. Have to look what is happen.
So far you should remove the parameter or rise the value.

tomhajjar, 2023-12-22:
The ngspice ver 41 manual is wrong on page 174 and 178 since it shows Nchan as shown below. What do you do for Pchan?

. MODEL IXTH48P20P VDMOS Pchan Vds =200 VTO =-4 KP =10 Lambda =5m...

LTSpice has the parameter "subthres" but ngspice doesn't show it as a parameter on page 175. ngspice flags it and ignores it.

dwarning, 2023-12-22:
No the line is different then yours in the schematic.
You should read my post as a whole: Put the Nchan (or Pchan) in to the braces and no hang in ngspice.
For the smal Ksubthres problem I opened a case in the bug tracker.

tomhajjar, 2023-12-22:
Qucs-S freezing Windows because ngspice misbehaves isn't good either. Makes it hard to diagnose the problem.

ra3xdh, 2023-12-22:
I cannot reproduce the freeze on Linux platform. Ngspice hangs, but it could be stopped with "Stop" button in the simulation dialog. Probably Ngspice freezes Windows itself.

ra3xdh, 2023-12-22:
Ngspice consumes all available RAM after some time. Nothing on Qucs-S side will help in this case.

tomhajjar, 2023-12-22:
@dwarning 

I figured out the source of the MRF101 model parameter being "subthres" instead of "ksubthres". LTSpice had it's own proprietary VDMOS model and a program called MOStool was used to create the model parameters. "subthres" was a parameter in the old version of the model. They now use "ksubthres". I guess I found an old MRF101 model.

Since many LDMOS/VDMOS devices don't come with Spice models, the tool can be used to make a simple model.

https://groups.io/g/LTspice/topic/vdmos_tool/36996094

dwarning, 2023-12-22:
Taking over my recommendations it works.
[image]

I know the LTspice model. We made the ngspice VDMOS compatible as possible but left out same parameter naming. And we have true self heating!
See ngspice/examples/vdmos

ra3xdh, 2023-12-25:
The reason of the crash was the wrong syntax of the VDMOS model. It is not possible to restrict the RAM consumption of the child process using the Qt tools. QProcess doesn't provide such mechanics. Closing this.

closed 2023-12-25 by ra3xdh

tomhajjar, 2023-12-30:
@dwarning

Holger did a fix for this. ngspice will allow both the "old" and new syntax. Don't know if the manual will be updated.

dwarning, 2023-12-31:
I see no conflict in the manual v.42. 
Anything is allowed:

`.model my_vmodel VDMOS pchan Vto=-4 Kp=4.7m
.model my_vmodel VDMOS (pchan Vto=-4 Kp=4.7m)
.model my_vmodel VDMOS pchan (Vto=-4 Kp=4.7m)
`