ra3xdh/qucs_s #405: Nutmeg Scripting for Power Amplifier Wingspread Analysis

tomhajjar, 2023-12-24:
I'm trying to recreate a PA "Wingspread" example supplied with ngspice in Qucs-S. The analysis is popular on the diyAudio forum using LTSpice so I recreated the ngspice example using LTSpice. LTSpice will plot the data using a single function, d(v(out)). ngspice seems to require a long script. I'm not having luck getting the ngspice script to work in Qucs-S.

[image]

[image]

[image]

[Audio_Output_Linearity_prj.zip](https://github.com/ra3xdh/qucs_s/files/13762882/Audio_Output_Linearity_prj.zip)

ra3xdh, 2023-12-25:
You don't need to use the script to perform this simulation. Everything could be made using the Par.Sweep and equations. The script was needed to emulate parameter sweep that is provided by means of GUI. Look at the attached example.
[image]

tomhajjar, 2023-12-25:
Thanks.

I'm still confused why the *.dat.ngspice data sets are blank. "Data" was being written to the file specified in the script.  I thought I followed what was done in Potentiometer_Nutmeg.sch.

ra3xdh, 2023-12-25:
> I'm still confused why the *.dat.ngspice data sets are blank.

You are trying to write the variable that is obtained in a loop. The vectors may be reinintialized after each loop operation. You have only to form the custom ouput with `write` and `appendwrite` here. Other combinations will not work.

On the other hands you need not to use the Par.Sweep emulation with loop and script here. I have implemented variable sweep a long time ago. See #189 The potentiometer device itself will be available since the next release See #122

ra3xdh, 2023-12-25:
Closed as resolved.

closed 2023-12-25 by ra3xdh