ra3xdh/qucs_s #406: QUCS-S 2.1.0: Tuner Slider Bug

3813127458, 2023-12-25:
Problem:

The tuner slider parameter do not update accordingly in case the Max, Min, Val, Step field values have been changed.
The slider parameters are not fully consistent with the parameter fields.

Problem Demo:

<<Tuner_Slider_Bug_Video.zip>>
<<HBM_9_prj.zip>>

Solution:

If any  Max, Min, Val, Step field values have been changed: update the tuner slider parameters and VAL if VAL is outside the MIN to MAX range.

[HBM_9_prj.zip](https://github.com/ra3xdh/qucs_s/files/13763461/HBM_9_prj.zip)
[Tuner_Slider_Bug_Video.zip](https://github.com/ra3xdh/qucs_s/files/13763463/Tuner_Slider_Bug_Video.zip)

ra3xdh, 2023-12-25:
This bug was fixed by https://github.com/ra3xdh/qucs_s/commit/d0b6e74a7aa76ea72cff5e7b8b0800f15430ba5a Here is the screencast illustrating how it looks in the version from Git. The fix will be available since the next release.
[image]

ra3xdh, 2023-12-26:
Closing as completed. Fixed on `current` branch.

closed 2023-12-27 by ra3xdh