ra3xdh/qucs_s #407: Backward schematic compatibility

tomhajjar, 2023-12-25:
When testing different revisions of "Qucs" one is forced to use a text editor to open a "new" schematic in older revisions of "Qucs".

I'm use to editing and converting "Qucs" files in a text editor but others may not be.

ra3xdh, 2023-12-25:
This is expected and done intentionally. The versioning has only forward compatibility. Qucs-S asks if the circuit could be opened from the newer version. For the most of cases everything should be loaded.  But you cannot open Qucs-S circuit with Qucs. The opposite is possible.

ra3xdh, 2023-12-25:
I have just checked this. I am able to open Qucs-S v24.1.0 schematic using Qucs-S v1.0.1 after pressing "Yes" in the dialog.

ra3xdh, 2023-12-25:
There exists "Load documents fro future version" setting. Set this option in the application settings dialog if you need to test different versions. This will load every schematic without version check.
[image]

closed 2023-12-28 by tomhajjar