ra3xdh/qucs_s #410: S-Parameter and Transient Simulation on Same Circuit Give Wrong Transient Results

3813127458, 2023-12-26:
It is useful to run e.g. transient simulation and S-parameter simulation on the same circuit.
This is done by connection of the S-Parameter port (=ac power source) and a voltage pulse source in series.

[Spar.zip](https://github.com/ra3xdh/qucs_s/files/13772944/Spar.zip)
[image]

Problem: 

The ac power source does not represent its impedance Z0 as real resistor.
Therefore the transient simulation gives a wrong result.

ra3xdh, 2023-12-26:
The result is not wrong. I am representing the S-par port as the sine source. The magnitude is calculated from `P` parameter.  This is needed to simulate both transient and S-par response in one simulation without additional source connection. For example it is useful for linear amplifier analysis. The Qucsator simulation also follows such approach. This representation is needed to provide the backward compatibility between simulators. So, I cannot replace the port by 50-Ohm resistors. It is not subject of change or subject of discussion.

3813127458, 2023-12-27:
Unfortunately the result is wrong for simultaneous S-parameter and transient simulation. 
The two circuits in the following screen shot must have same transient result, which is not the case.

Potential solution: create new S-parameter port using correct math for all domains.
The existing ac source ensures backward compatibility.

[image]

ra3xdh, 2023-12-27:
The simulation result is correct. Port is a sine source for transient. It provides frequency and magnitude defined as port parameters. Qucsator represents the port as the sine source for transient too. I can provide an options to disable the transient model. This will not affect the backward compatibility. The transient model will be enabled by default. 

You may define ports manually using the SPICE syntax and generic voltage source that allows full SPICE string. The simulation will work. Refer to section 4.1.11 of the Ngspice manual. Use this device:

[image]

ra3xdh, 2023-12-27:
I have added TODO list and changed the issue title.

ra3xdh, 2023-12-27:
I have added an `EnableTran` option for RF port. This option is active by default. It `EnableTran=false` the transient model for RF port is disabled. The port is represented as `dc 0` for transient analysis. 

[image]

closed 2023-12-27 by ra3xdh

3813127458, 2023-12-27:
Excellent! - the final example RTF_N7 works perfect now simultaneous S-par and transient ... thx

[RTF_N7.zip](https://github.com/ra3xdh/qucs_s/files/13779842/RTF_N7.zip)
[image]

tomhajjar, 2023-12-28:
Network for a distributed amp?

3813127458, 2023-12-29:
It is a simplified equivalent model of a lossy transmission line.

It can be used as a low pass filter which is matched in the pass band AND stop band simultaneous, which is not the case for classical low pass filters. Other application is pulse shaping in the time domain.

References:

Djordjevic et al. "On a Class of Low-Reflection Transmission-Line Quasi-Gaussian Low-Pass Filters and Their Lumped-Element Approximnationns" in IEEE Trans. on Microwa ve TI eryniad Techniques, vol. 51 of of 7, pp. 18711-877, July 2003.

J. Breitbarth and D. Schmelzer "Absorptive Near-Gaussian Low Pass Filter Desigfn with Applications in the Time and Frequency Domain" in IEEE Microwave Symposium Digest, MTT-S International, vol. 3, pp. 1303-1306, Jun 2004.

tomhajjar, 2023-12-29:
Thank you for the explanation. It reminded me of a distributed amplifier.

commit 7df0e3e9 (2023-12-27): Add EnableTran option for RF port; fixes #410