ra3xdh/qucs_s #411: Add entry field for external model for Verilog-A

dwarning, 2023-12-26:
Vadim,
I have a big wish.
The life would be much easier if there would an entry field for model cards which can provided by include a file. I made a sketch:
[image]
At the moment I have to change the parameters painful by editing the parameter entries. If I change the model anything is lost and we have the fallback to the Verilog-A default values.
My proposal is to add a checkbox and an entry field for a model name. If both exist the user have to provide the model card in a file which has to be included. In this case the modelcard from symbol have to omitted in the netlist.

dwarning, 2023-12-31:
No response so far. For me a show stopper because I see no chance to include model cards for VA models.
I will close.

ra3xdh, 2024-01-01:
> No response so far. 

The response may require some time, because I am developing Qucs-S in my spare time. Usually I am responding within 1 week, but some issues may require longer time. The project has no full-time developers. Please wait. 

> if there would an entry field for model cards which can provided by include a file

The implementation of your proposal is not so easy as it seems. The current design of verilog-a device is strongly dependent on JSON files. The symbol is separate JSON object. I would not touch this software solution, but rewrite it from the scratch. The adding of the modelcard reference may be not possible using the existing implementation. I am planning to implement Verilog-A file device that will act like "SPICE file" device and will not require the usage of JSON.  But the implementation of this feature has a low priority for me because of low significance of compact modelling for 99% of average Qucs-S users.  So, don't expect the new Verilog-A device will be added in this year.

But there exist a workaround for your request using the existing  "SPICE generic device component". You may aslos attach one model to many devices (for example N or P type) using this workaround. Perform the following steps:

* Compile the `*.va` file into the OSDI module. No matter if form Qucs-S of from CLI.
* Put "SPICE generic (Z)" device on schematic. Define the letter "N" and pin number. Fill the model reference field.
* Put ".MODEL" modelcard device on schematic. Fill the modelcard in it.   
* Put "SPICEINIT" device on schematic. Fill the "osdi <path_to_osdi>" command here. 
* Add simulations and other devices. Pack the "SPICE generic device" into subcircuit if necessary. Define symbol and forward subcircuit parameters into the modelcard. 

The attached screenshots illustrate this workaround for tunnel diode model.

[image]
[image]
[image]

ra3xdh, 2024-01-01:
Here is an example project for the proposed workaround.

[Tunnel.zip](https://github.com/ra3xdh/qucs_s/files/13804699/Tunnel.zip)

dwarning, 2024-01-02:
Thanks Vadim,
this is a acceptable solution with positive side effects: I don't need load the Verilog-A module before (the automatic load is still missing) and it is a solution for the both channel types (e.g. nch/pch) from one .va file. No need to copy!
What I am missing is that the subcircuit filename is not shown as default. I have to switch it on anytime I placed the subcircuit in to schematic.

ra3xdh, 2024-01-02:
Closing as resolved.

closed 2024-01-02 by ra3xdh