ra3xdh/qucs_s #415: False error reported?

tomhajjar, 2023-12-28:
An error was reported during simulation but the Log only shows the term "error_amp_out" which isn't an actual error.

[image]

[LM337_TRANS_ngpice_prj.zip](https://github.com/ra3xdh/qucs_s/files/13782612/LM337_TRANS_ngpice_prj.zip)

ra3xdh, 2023-12-29:
Yes, Qucs-S shows "Simulation error" message if the log contain the string "error". It is not obvious how to fix it, because Ngspice has no unified format for the error message. Maybe some RegExp would help.

dwarning, 2023-12-29:
I am not sure and don't know the model but I think error_amp_out is a node name - the output of the regulator error amplifier. So don't worry - no ngspice problem.

tomhajjar, 2023-12-29:
Yes the "error" is false. 

Unfortunately after I GREP'd the source I found dozens of different "Error" text strings. It would be hard to parse so many variations and the number only increases every release.

ra3xdh, 2023-12-29:
@dwarning , The solution may be to flag "Simulation error" in the Qucs-S only if the log contains "Error" or "ERROR" strings. The "error" in lower case most probably means the node name.

tomhajjar, 2023-12-29:
I searched the ngspice forum and GREP'd the source and found the following error strings. There are many more...

Error:
Error(
Error (
Error on line XXX:
Error on line:
Syntax error:
Expression err:
Numparam expansion errors:
Error in netlist line number...

ra3xdh, 2024-01-06:
I have added **TODO** list for this issue.

ra3xdh, 2024-01-07:
I have implemented pattern recognition for error messages. The node name error should not raise error flag anymore. Closing this as completed.

closed 2024-01-07 by ra3xdh