ra3xdh/qucs_s #416: Qucs-S 24.1-7df03e9 -  Yet Another Tuner Slider Bug

3813127458, 2023-12-28:
Probably this little bug is related to system language (EN) and system locales (DE) combination(?):

[Tuner_Slider_Bug.zip](https://github.com/ra3xdh/qucs_s/files/13783886/Tuner_Slider_Bug.zip)
[Screencast from 2023-12-28 07-06-47.webm](https://github.com/ra3xdh/qucs_s/assets/75557552/b0b7d6cc-f17e-4b3c-b414-ce78ea794253)

ra3xdh, 2023-12-28:
Yes, it seems the parser cannot recognize the decimal delimiter for DE locale. I will look into this.

ra3xdh, 2024-01-05:
@3813127458 Could you tell your `locale` settings? I have tried with `de_DE` locale and the behavior is a bit different.

3813127458, 2024-01-06:
The `locale`  command gives the following listing:      
                                                                                                                                                              
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=de_DE.UTF-8
LC_NAME=de_DE.UTF-8
LC_ADDRESS=de_DE.UTF-8
LC_TELEPHONE=de_DE.UTF-8
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION=de_DE.UTF-8
LC_ALL=

On the actual development branch Qucs-S 24.1-9fc4db38 the problem still exists.

I use the following desktop:

Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.10
KDE Frameworks Version: 5.113.0
Qt Version: 5.15.11
Kernel Version: 6.6.8-2-MANJARO (64-bit)
Graphics Platform: X11

3813127458, 2024-01-09:
If you have a fix I will test it.
The tuner slider is a key feature ... should work very solid in the next major release.

We should think also about two further issues (optional):

1) Tuner limits and step size saved optional in parameters `VAL[[[:MIN]:MAX]:STEP]`.
This is useful in many cases, as it is not required to re-define the tuner slider parameters every time again.
Could be used also for optimizer parameter limits.

2)  Simulation time: if a circuit becomes more complex (>1 sec simulation time), the tuner slider becomes more or less useless. There are several options how to handle this.
a) let the tuner slider free moving and hand over the actual value if the simulator stops by
b) start multiple simulator kernel threads

ra3xdh, 2024-01-10:
I have tried to use your locale settings, but I cannot reproduce this issue. The numbers use the syntax with dot on my machine and tuner dialog has no problems. I have tried Ubunutu-22.04 with Qt 5.15 and Qt 6.2.4. I suspect this may be Qt version dependent or platform dependent.

3813127458, 2024-01-12:
I have changed from Manjaro KDE/X11 to Manjaro GNOME/Wayland desktop and faces same issue:

[Screencast_from_2024_01_12_15_58_18.webm](https://github.com/ra3xdh/qucs_s/assets/75557552/6189e1d0-e370-4cb0-91e3-33c8092cc96f)

locale:
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=de_DE.UTF-8
LC_NAME=de_DE.UTF-8
LC_ADDRESS=de_DE.UTF-8
LC_TELEPHONE=de_DE.UTF-8
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION=de_DE.UTF-8
LC_ALL=

System:

# System Details Report
---

## Report details
- **Date generated:**                              2024-01-12 16:10:16

## Hardware Information:
- **Hardware Model:**                              ASUSTeK COMPUTER INC. Vivobook Go E1504FA_E1504FA
- **Memory:**                                      8,0 GiB
- **Processor:**                                   AMD Ryzen™ 5 7520U with Radeon™ Graphics × 8
- **Graphics:**                                    AMD Radeon™ Graphics
- **Disk Capacity:**                               (null)

## Software Information:
- **Firmware Version:**                            E1504FA.306
- **OS Name:**                                     Manjaro Linux
- **OS Build:**                                    rolling
- **OS Type:**                                     64-bit
- **GNOME Version:**                               45.2
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.6.8-2-MANJARO

ra3xdh, 2024-01-12:
What Qt version are you using? Is it Qt5 or Qt6 build?

3813127458, 2024-01-12:
Qucs-S `About Qt` log: Qt 6.6.1

I always use these commands to build the current branch:

```
git clone https://github.com/ra3xdh/qucs_s.git
cd qucs_s
mkdir builddir
cd builddir
cmake ..  
make
sudo make install
```

3813127458, 2024-01-22:
Also on **openSUSE Tumbleweed** the dot is not accepted in the tuner slider input fields:

[issue.webm](https://github.com/ra3xdh/qucs_s/assets/75557552/3bda64c6-efbb-4c60-bfa8-b2b386481177)

[HBM_SM_30.sch.zip](https://github.com/ra3xdh/qucs_s/files/14006001/HBM_SM_30.sch.zip)

Operating System: openSUSE Tumbleweed 20240119
KDE Plasma Version: 5.27.10
KDE Frameworks Version: 5.114.0
Qt Version: 5.15.12
Kernel Version: 6.6.11-1-default (64-bit)
Graphics Platform: X11
Processors: 4 × Intel® Core™ i7-4600U CPU @ 2.10GHz
Memory: 15,3 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 4400
Manufacturer: CompuLab Ltd.
Product Name: Intense-PC2 (IPC2)
System Version: 1.x

LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=

KDE keyboard numblock setting:

[image]

ra3xdh, 2024-01-22:
I still cannot reproduce it on Ubuntu. I have to install one of the mentioned distributions in the VM to make debugging. Don't expect a quick fix by this reason.

wawuwo, 2024-02-28:
@3813127458 on the video from the original post you type in 1.0 which is then parsed as 10 (1,00E+01) because of the German locale. If I'm not mixing it all up German uses points to delimit thousands and in other places points mean nothing. 10 > 2.9095 → min value is invalid.

But had you used comma to delimit fractional part from integer one, it wouldn't have worked anyway, because [text-to-float conversion](https://github.com/ra3xdh/qucs_s/blob/edc6bd8f610741e2f93398f407216ce4d68d697f/qucs/dialogs/tuner.cpp#L598) is locale [unaware](https://doc.qt.io/qt-5/qstring.html#toFloat)

So I believe the _actual_ bug is the text-to-float conversion in tuner dialog.

I can try to prepare a fix, but I can't say how long will it take, spare time has been scarce lately :(

3813127458, 2024-02-29:
It seems the bug has been fixed by someone (maybe Vadim?) a couple of weeks ago.

I am actually testing Windows version 24.1-afe87e7b and Linux version 24.2 on openSUSE Tumbleweed and can confirm that the tuner is working seamless without any issues with my locales de setting, both Windows and Linux.

Therefore, I think this issue could be closed for now.

wawuwo, 2024-02-29:
Hm, that's strange. What's your locale settings on Tumbleweed?

3813127458, 2024-02-29:
werner@ipc2:~> locale
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=

Operating System: openSUSE Tumbleweed 20240227
KDE Plasma Version: 5.27.10
KDE Frameworks Version: 5.115.0
Qt Version: 5.15.12
Kernel Version: 6.7.6-1-default (64-bit)
Graphics Platform: X11
Processors: 4 × Intel® Core™ i7-4600U CPU @ 2.10GHz
Memory: 15,3 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 4400
Manufacturer: CompuLab Ltd.
Product Name: Intense-PC2 (IPC2)
System Version: 1.x

wawuwo, 2024-02-29:
I've tried Qucs-S from the `current` branch:
- with `LC_NUMERIC=ru_RU.UTF-8` it's not possible to input points (that's OK), but values with fractional parts (i.e. with comma) raise error message
- With `LC_NUMERIC=de_DE.UTF-8` I can input any values with point, but the slider seems to ignore them.

@3813127458 could you please check whether the range of the slider actually changes when you input new values to "Min" or "Max" fields in tuner dialog?

3813127458, 2024-02-29:
Yes, you are right, the bug at the slider limits still exists:

[image]

[HBM_9_prj.zip](https://github.com/ra3xdh/qucs_s/files/14444828/HBM_9_prj.zip)

I think in general the numbers should be point separated. Usually I do no take care about a locale setting and use always point "." as a separator in all scientific numeric applications.

wawuwo, 2024-03-09:
@3813127458 I've made a fix base on your idea of using dot as the only possible separator in numbers, PR #636

ra3xdh, 2024-03-11:
Fixed via #636

closed 2024-03-11 by ra3xdh