ra3xdh/qucs_s #418: Tuner Slider GUI Window Size Scaling Issue

3813127458, 2023-12-29:
Qucs-S 24.1-7df0e3e9
Linux 6.6.8-2-MANJARO
GNOME 45.2
Wayland

The tuner slider GUI window shows uncomfortable window size auto scaling.

Potential solution: pre-asses parameters field length  and change GUI window size only if it is necessary.
The solution need to be tested with e.g. 10 tuner slider in operation parallel.

[Tuner_Slider_GUI_Window_Issue.webm](https://github.com/ra3xdh/qucs_s/assets/75557552/71a75b3e-f092-43ee-b82a-9bb9203539b3)

[RTF_N7_master.zip](https://github.com/ra3xdh/qucs_s/files/13791424/RTF_N7_master.zip)

ra3xdh, 2023-12-29:
I cannot reproduce these issues using X11/KDE or Windows. Need to wait for someone who can debug and provide a patch for Wayland.

3813127458, 2024-01-06:
I did test Debian 12, Gnome Desktop on Wayland, and it works without any problems.
Therefore, it is probably a specific problem of the actual Manjaro/Gnome/Wayland desktop.

ra3xdh, 2024-05-23:
I have tested the Wayland+Qt6.7.0 platform. I still cannot reproduce exactly the same issue, but the other tuner window size issues appear time to time. It's starnge, but the Windows build using the same Qt version is not affected.

ra3xdh, 2024-06-02:
@3813127458 I still cannot reproduce the bug exactly as on your screen recording, but another tuner window size issues appear on my system. Do you see the behavior as in the attached screencast? Only Wayland is affected. The X11 and Windows work as expected using Qt6. 
[Screencast_20240602_085730.webm](https://github.com/ra3xdh/qucs_s/assets/4920080/c4dcdadd-bcd2-45ca-8acf-2da2d6ac9eef)

ra3xdh, 2024-06-04:
It seems the `QWidge::adjustSize()` (`tuner.cpp` line 749) doesn't work properly for Qt6 using Wayland backend. Only hardcoded minimum size for `TunerElement` helps. I failed to provide a fix for this issue.

ra3xdh, 2024-06-09:
The following solution for these issues is possible:

* Hardcode size hint for `TunerElement` class. This makes the tuner dialog autoscaled properly on Wayland
* Wait for the next Qt6 version. There exist probability that this issue is related to some bugs in QtWayland backend, because X11 and Windows works properly. 

If this issue will be not self fixed after the next Qt6 update until the Qucs-S release (planned for September), I will accept the second solution.

3813127458, 2024-06-09:
Maybe this was a specific issue of MANJARO / GNOME 45.2 / Wayland.
At the moment I have no MANJARO / GNOME / Wayland platform available for testing.

The latest Opensuse / Tumbleweed / KDE / Wayland and X11 works fine.

ra3xdh, 2024-06-09:
KDE6+Manjaro+Wayland has not exactly the issue from the first post but this issue instead: https://github.com/ra3xdh/qucs_s/assets/4920080/3a27311c-0bdf-404d-b77f-8a9d61a9bc39 Could you reproduce this using OpenSUSE+KDE+Wayland?

3813127458, 2024-06-09:
Yes, I can confirm, that the cursor goes to window scaling mode after selecting a value to be tuned:

https://github.com/ra3xdh/qucs_s/assets/75557552/62141e07-f3ce-4e36-a4b9-908346a8cd61

Qucs-S 24.2.0 

Operating System: openSUSE Tumbleweed 20240607
KDE Plasma Version: 6.0.5
KDE Frameworks Version: 6.2.0
Qt Version: 6.7.1
Kernel Version: 6.9.3-1-default (64-bit)
Graphics Platform: Wayland
Processors: 4 × Intel® Core™ i7-4600U CPU @ 2.10GHz
Memory: 15,3 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 4400

ra3xdh, 2024-07-23:
The most likely it is Qt/Wayland issue. I am observing the window scaling issues with other application like KiCAD on my system. It is unclear if it is needed to add explicit fix for this. Closing this.

closed 2024-07-23 by ra3xdh