ra3xdh/qucs_s #421: Loudspeaker model for Qucs-S

tomhajjar, 2023-12-29:
I made a loudspeaker library for amplifier testing. It is based on an article popular at diyAudio. 

First entry is a 2-way 8-Ohm. It presents a real-life reactive load unlike resistor load.

[image]

[image]

[Loudspeaker_prj.zip](https://github.com/ra3xdh/qucs_s/files/13794980/Loudspeaker_prj.zip)

ra3xdh, 2023-12-29:
Thanks for the contribution! I am planning to add a loadspeaker library in the next release.

tomhajjar, 2023-12-29:
diyAudio has a number of models for "old school" loudspeakers from the 60s/70s. 

Qucs also had a number of loudspeaker and enclosure projects.

[Loudspeakers_qucs.zip](https://github.com/ra3xdh/qucs_s/files/13795123/Loudspeakers_qucs.zip)

tomhajjar, 2024-01-19:
Holger modified an LTspice loudspeaker library to work with ngspice. The library has individual loudspeakers for use in designing crossover networks.

[Loudspeaker_lib.zip](https://github.com/ra3xdh/qucs_s/files/13992340/Loudspeaker_lib.zip)

csrabak, 2024-01-25:
Just a rambling, but perhaps could we have a "component" _speaker_, which would show with its distinctive symbol with two parameter resistance and inductance?

tomhajjar, 2024-01-26:
Your welcome to make submissions. In general a speaker model should either reflect a real life 2/3 way speaker or popular individual woofers/midranges/tweeters. Impedance should be nominally 4/8 Ohms.

Even a simple speaker cannot be modeled with an inductor and resistor. You need multiple resonances and realistic Impedance.

csrabak, 2024-01-26:
Humm I'm affraid I could not communicate what I was thinking...

Whereas for big systems a "loudspeaker box" is more properly modeled by the crossover network plus the speakers as ultimate loads that dissipate the energy and convert it to sound waves, I was thinking in having these last ones becoming a specific component, wich could allow _even_ to have a propor symbol in the schematic.

Otherwise not all circuits which use speakers come from the [passive] crossover networks in the power path, in fact, there is a trend where the frequency separation is done in low level and separate power amplifiers feed the different speakers[^1].

Also for other applications, say the audio part of a cell phone, a portable TV set, the speaker is the load of the audio amplifier.

I do agree that a more complete transducer model would require a modeling of the conversion to the mechanical sound waves and require an additional pin in the symbol which could be sampled for ploting, etc.

I was not thinking in this level of sophistication, as the sound pressure response would be dependent of the mounting of the transducer (that's why we put these objects in boxes, after of all!).

[^1]: One I can remember is described in the AN TIDU035 from TI.

ra3xdh, 2024-01-29:
I am planning to add a speaker model using this approach: https://circuitdigest.com/electronic-circuits/simulate-speaker-with-equivalent-rlc-circuit I will add a new library soon.

ra3xdh, 2024-02-08:
#535 implements the loudspeaker library.
[image]

closed 2024-02-08 by ra3xdh