ra3xdh/qucs_s #423: ngspice % done readout

tomhajjar, 2023-12-30:
In porting an old QucsStudio HICUM project over to ngspice I ran into issues.

1. ngspice crashed. Holger recommended adding "uic" to get around the problem. 

https://sourceforge.net/p/ngspice/discussion/120973/thread/56801dcb2b/

2. Once uic was added, ngspice analysis using the built-in HICUM model is so slow I though something was wrong. DuSpice gives a percent done indication which helps. Can this be added to Qucs-S?

3. I noticed that using .OP (F8) to check bias levels causes the pulsed source Vref to be 0 volts regardless of changing settings that would "should" make it -5 volts initially. It meant I couldn't check bias unless I used a fixed source in it's place.

Ignore the plot differences. I assume the Ft of the 2 HICUM devices is very different. 

[image]

[image]

[image]

[HICUM2_CML_Oscillator_prj.zip](https://github.com/ra3xdh/qucs_s/files/13798405/HICUM2_CML_Oscillator_prj.zip)

ra3xdh, 2024-01-06:
> There is something wrong that slows down the Qucs-S->ngspice interaction.

I have executed the netlist in terminal and obtained the same behavior. Simulation is also slow. Qucs-S doesn't slow down the simulation or applay additional options. Please check the `spiceinit` used by DuSpice. I suspect the different paths to HICUM models or even different ngspice version used by Qucs-S and DuSpice.  It is not Qucs-S error. 

> DuSpice gives a percent done indication which indicates it hasn't hung or crashed.

I have created some long TRAN simulations and tried to execute it. The Qucs-S GUI updates the log window dynamically that indicates the simulation is running. Ngspice doesn't report the percentage. I didn't find a CLI option to enable the percentage report in Ngspice. So I cannot add it to the Qucs-S GUI. I don't know how DuSpice derives percentage is the log doesn't report it. Qucs-S cannot show percentage if the log doesn't contain it. 

> I noticed that using .OP (F8) to check bias levels causes the pulsed source Vref to be 0 volts 

This is done intentionally. All sources except DC has the option `dc 0`. Use SPICE voltage source (the red one) the allows to define the full SPICE string if you need different behavior.

closed 2024-01-06 by ra3xdh

tomhajjar, 2024-01-09:
Waiting for ngspice guys.
https://sourceforge.net/p/ngspice/discussion/133842/thread/46601c27e9/

 I Grep'd the DuSpice source but it wasn't obvious what to look for.
https://ngspice.sourceforge.io/experimental/ng_gui_sources.7z