ra3xdh/qucs_s #425: Angelov HEMT model

dwarning, 2023-12-31:
Attached is a qucs-s project with the Angelov HEMT model written in Verilog-A from [VA-Models](https://github.com/dwarning/VA-Models). All included test cases working with openVAF and ngspice version > 41. The model includes all basic characteristics and additional self heating effect.
Problems in qucs-s:
- No possibility include external .model cards for Verilog-A models
- Few times the plotting capabilities are limited, especially dc-sweep over ac and sp analysis
- Resistors can't get noisy=0 instance parameter
- The default generated lines generate some waste in working directory:
`echo "" > spice4qucs.cir.noise
echo "" > spice4qucs.cir.pz
`

[image]

[image]

[image]

[image]

[image]

[Verilog-A_Angelov_prj.zip](https://github.com/ra3xdh/qucs_s/files/13800802/Verilog-A_Angelov_prj.zip)

tomhajjar, 2023-12-31:
Nice project. Should I be getting "no such parameter temp" errors? 

[image]

[image]

dwarning, 2023-12-31:
> Nice project. One of the few "Qucs-S" Verilog projects that works on my system.

Thank you

> 
> Should I be getting "no such parameter temp" errors?

Don't know why it happen on your side. Will check it with Windows next year (!) Perhaps Temp should changed to devTemp or similar,

> 
> For future reference, Qucs won't let you have a single Verilog.va file for both "n" and "p" devices. You need to make separate "n" and "p" files.

That is another reason to stop with qucs-s.

Happy New Year.

ra3xdh, 2024-01-01:
> No possibility include external .model cards for Verilog-A models

See the workaround in #411 

> Few times the plotting capabilities are limited, especially dc-sweep over ac and sp analysis

Please add more explanations on this. What exactly is limited?

> Resistors can't get noisy=0 instance parameter

Probably the red resisitor may help you. These resisitors allows the full SPICE string including model.

> The default generated lines generate some waste in working directory:

It is not a waste. The netlister uses this workaround to remove possible old NOISE and PZ datasets. Furthermore, the user must not to look into the cache directory. I don't see the problem here.

ra3xdh, 2024-01-06:
Assume the questions are explained here and in #411. Closing this.

closed 2024-01-06 by ra3xdh