ra3xdh/qucs_s #426: Schematic Editor: View All -> wrong scaling

3813127458, 2024-01-01:
The schematic editor performs wrong graphics scaling of the schematic diagram at "View All".
Probably the long Nutmeg script is the root cause for this issue(?)

[HBM_9.sch.zip](https://github.com/ra3xdh/qucs_s/files/13804121/HBM_9.sch.zip)

https://github.com/ra3xdh/qucs_s/assets/75557552/ced6cdc9-9eb2-4f92-81f5-7538785205f7

Version Info:

Qucs-S 24.1-7df0e3e9
Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.10
KDE Frameworks Version: 5.113.0
Qt Version: 5.15.11
Kernel Version: 6.6.8-2-MANJARO (64-bit)
Graphics Platform: X11
Processors: 4 × Intel® Core™ i7-4600U CPU @ 2.10GHz
Memory: 15,3 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 4400
Manufacturer: CompuLab Ltd.
Product Name: Intense-PC2 (IPC2)
System Version: 1.x

wawuwo, 2024-01-01:
I've conducted a quick research: it seems like the cause of described behavior is [this line](https://github.com/ra3xdh/qucs_s/blob/7df0e3e97b0bb4ea353d5a4f60cea0b2989232ac/qucs/components/component.cpp#L98). It calculates the width of a rectangle able to hold the text of the script, but selected `boundingRect` method doesn't treat newlines as line breaks (see QT doc [there](https://doc.qt.io/qt-5/qfontmetrics.html#boundingRect-1), so the script is treated as a _one-liner_. Possible solution is to use one of the overloaded `boundingRect` methods which treat newlines properly. 

I'll try to make a fix using a proper method.

ra3xdh, 2024-01-03:
Fixed by #429

closed 2024-01-03 by ra3xdh

tomhajjar, 2024-01-06:
Another way this issue is caused is when a wire or any "Painting" like a line, ellipse, circle..etc is so short or small to be unviewable. 

A "phantom" net caused by cut-paste or moving multiple items can also be a pain to locate and delete

I've had to use a text editor to find and delete the item...

wawuwo, 2024-01-07:
@tomhajjar could you please provide an example file or a video? I'm not sure I understand clearly how to reproduce the described issue.

tomhajjar, 2024-01-07:
Make a wire or any "Painting" like a line, ellipse, circle..etc is so short or small to be unviewable and place it out of the desired viewable area.

The "phantom" net is harder to reproduce since I don't know what causes it other than it randomly occurs during a cut-paste or moving of multiple items. The net name is visible but detached from the actual net itself. The net name can be so far away from the "schematic" it becomes invisible when zoom all is used. If you can find it you can click on it and delete it. I usually end up deleting the net and hope it disappears but unfortunately it doesn't always work. I end up using a text editor to find it in the schematic and delete it.