ra3xdh/qucs_s #432: Qucs-S creating corrupt libraries

tomhajjar, 2024-01-03:
I'm having issues with Qucs creating libraries consisting of only the symbol section. The Spice section is missing. The goal was to make a OpAmp library that worked for qucsator and ngspice. 

The problem seems to occur after jumping back and forth between using ngspice and qucsator multiple times to check that each model.sch for the library is functional for both qucsator and ngspice. 

What is really odd is once the "problem" happens, it seems like the project or files become corrupted. Every attempt fails. I cannot find anything wrong with the project or the files. Restarting Qucs doesn't help.  I even made a new project containing only the model.sch and models.cir files. The library was good only the first time. The name I used is OpAmp.lib. 

Not being able to overwrite a library even in the usr_lib folder is annoying...

**BAD**
[image]

**GOOD**
[image]

[Op-Amp_Library_prj.zip](https://github.com/ra3xdh/qucs_s/files/13823963/Op-Amp_Library_prj.zip)

tomhajjar, 2024-01-04:
Vadim

Can you explain what the "new" folders containing *.lst files in the usr_lib directory that are created when making the new library are used for? Qucs/QucsStudio never created them.

Are they needed by Qucs-S for qucsator or ngspice analysis or are they temporary? Can they be deleted after the library is made?

[image]

[image]

ra3xdh, 2024-01-04:
> The goal was to make a OpAmp library that worked for qucsator and ngspice. The name I used is OpAmp.lib.

The SpiceOpamp.lib doesn't contain qucsator models intentionally. The making of the complex IC library containing both SPICE/Qucsator netlist doesn't cost its time effort. The Qucsator has very poor performance for the netlist converted form SPICE. Also Qucsator has poor performance for schematics with many nonlinear devices. I have tried to make such library about 5 years ago, and it the simulation may hang or fail to converge even for simple Opamp circuit. So I decided to restrict realistic Opamp models by SPICE only. It's better to have no qucsator model that have the model that will fail to converge every time. The Qucsator should be restricted by only RF devices. Ideal Opamp is sufficient for Qucsator. 

>  Can you explain what the "new" folders containing *.lst files

These files are produced by `qucsconv` utility that converts SPICE netlists to Qucsator. The conversion is not performed when SPICE kernel is selected. The library contains the lst file reference instead of model inclusion. I don't know why Qucs developers decided to use *.lst files.  It's need to completely disable this procedure, because the may be misleaded. 

> Can Qucs-S make qucsator "compatible" libraries?

Yes, it can if the netlist contain only primitive devices and unified equations. For example the Crystal.lib was made in a such way.

ra3xdh, 2024-01-04:
I have added TODO for this issue. Outcome: conversion SPICE/Qucs should be disabled to prevent convergence errors or similar issues with qucsator. The qucsator model should be generated only for subcircuits that consists of equatios and primitive devices. Add warnings like "Subcircuit contains SpiceFile device. No Qucsaator model generated."

tomhajjar, 2024-01-04:
Vadim

What about the OpAmp Library missing the device Spice file information as shown in the "BAD" example? Were you able to recreate this?

ra3xdh, 2024-01-04:
> What about the OpAmp Library missing the device Spice file information as shown in the "BAD" example?

This library was created by Qucsator mode. It is activated when Qucsator is selected as the default simulation backend. This mode was intended to create libraries for pure Qucs without SPICE extensions. I would disable this mode and keep only Qucs-S library workflow.

tomhajjar, 2024-01-04:
Being able to create a library with the same name as one in the system or usr_lib folders without having to delete any first would be nice. Since the file is written to the usr_lib folder a warning should be sufficient before allowing over-write. System libraries can always be replaced if deleted or over-written.

tomhajjar, 2024-01-04:
As a side note, the symbols used in the original OpAmps.lib and SpiceOpamp.lib have issues. That was another reason for creating a new OpAmp library. The symbols used in OpAmps.lib SpiceOpamp.lib don't indicate Vcc, Vee or GND/Enable. When you mirror the device it is easy to connect the device wrong. I propose using symbols with labeling for each pin. 

I also did work on making a Voltage Comparator library since Qucs-S lacks one.

[image]

ra3xdh, 2024-01-04:
> the original OpAmps.lib and SpiceOpamp.lib have issues

I would edit the library in the text editor and add labels near the pins instead of recreation of the library. You may edit one symbol and look its code to determine where to insert label definitions. 

>  the original OpAmps.lib 

This library is deprecated and kept only compatibility with old schematics. It should not be modified. See #232 

> I also did work on making a Voltage Comparator library since Qucs-S lacks one.

Yes, this contribution is highly welcome. The `SpiceOpamp` library contains only LM311 model.

tomhajjar, 2024-01-04:
Here is the revised SpiceOpamp.lib and test data to confirm the labels assigned to pins are correct. SpiceOpamp.lib is in the folder.

[image]

[image]

[image]

[image]

[SpiceOpamp_library_prj.zip](https://github.com/ra3xdh/qucs_s/files/13833304/SpiceOpamp_library_prj.zip)

ra3xdh, 2024-01-04:
> Here is the revised SpiceOpamp.lib 

Thanks! I have just uploaded an updated library in the repository.

ra3xdh, 2024-02-01:
Fixed via #520

commit 8cf7f5c5 (2024-02-01): Merge pull request #520 from ra3xdh/fix_421

Fix #432 library issues
closed 2024-02-01 by ra3xdh