ra3xdh/qucs_s #433: Regulators.lib modified

tomhajjar, 2024-01-04:
I modified Regulators.lib to use "REG" as the Symbol ID versus the original way done in Qucs where the Symbol ID was the regulator part number which is redundant. Regulators.lib is in all the projects.

Old:
<Symbol>
  <.ID -40 -66 LM317_>

New:
<Symbol>
  <.ID -40 -66 REG>

[Voltage_Regulators.zip](https://github.com/ra3xdh/qucs_s/files/13835950/Voltage_Regulators.zip)

ra3xdh, 2024-01-08:
Thanks! I have just uploaded a new `VoltageRegulators.lib` based on your project. I have modified the LM1084, LM317, and LM337 models to provide Ngspice compatibility. These models doesn't require LTspice compatibility mode anymore. Closing this as completed.

ra3xdh, 2024-01-08:
The TL431 model will go to the new voltage references library. See #439

closed 2024-01-08 by ra3xdh

tomhajjar, 2024-01-08:
Are you going to blacklist Regulators.lib for ngspice?