ra3xdh/qucs_s #436: New Voltage Comparator Library

tomhajjar, 2024-01-07:
I'm slowly making progress on a new Voltage Comparator Library called Comparators.lib. Biggest hurdle is issues in ngspice. Holger and the crew are slowly fixing some of the issues. The two threads below outline some of the issues.

https://sourceforge.net/p/ngspice/discussion/120973/thread/44c6f0ae09/
https://sourceforge.net/p/ngspice/discussion/120973/thread/f6970fa18a/

It turns out most Linear Tech device Spice models use LTSpice functions not supported by ngspice. That isn't a big issue but the Spice models from TI and Analog Devices for devices compatible with the LT ones also don't work. Hopefully future fixes to ngspice will improve compatibility with manufacturers models.

On a side note, the TI "transient" models seem to produce more accurate data than older models. The LM311 model used in OpAmps.lib and SpiceOpamp.lib gives erroneous data when the input Common Mode range is exceeded. The model LM311TI.cir works better. I will use both to maintain backward compatibility.

Included is a new symbols_OpAmp.sch to ease making new devices.

[image]

[Voltage_Comparators_prj.zip](https://github.com/ra3xdh/qucs_s/files/13853931/Voltage_Comparators_prj.zip)

tomhajjar, 2024-01-09:
Waiting for ngspice guys. Once the fixes are in for "Tlines" lot's of TI "behavioral Spice models should work., 

https://sourceforge.net/p/ngspice/discussion/120973/thread/44c6f0ae09/

tomhajjar, 2024-01-12:
First cut at the VoltageComparators.lib.

I used your LM311 model in SpiceOpamp.lib. Same as used in LTSpice. I also added LM311TI transient model.

Waiting for feedback from ngspice so TLV3501 and TLV3601 can be added but TLV3603 is similar and faster.

Bothers me that some comparators like AD8561 and AD8611 don't simulate unless ".OPTIONS rshunt=1G" is added per Holger's recommendation. I added it to the TLV3501 and TLV3601 without luck. Still something wrong with ngspice and "Tlines".

[image]

[Voltage_Comparators.zip](https://github.com/ra3xdh/qucs_s/files/13910720/Voltage_Comparators.zip)

tomhajjar, 2024-01-14:
Update to add TLV3501 and TLV3601. ngspice must be the latest release for all the devices to work.

[Voltage_Comparators.zip](https://github.com/ra3xdh/qucs_s/files/13931915/Voltage_Comparators.zip)

ra3xdh, 2024-01-14:
Looks impressive! Are you planning to add more devices? I will upload the library in the repository once it will be finished.

tomhajjar, 2024-01-14:
For now no. It depends on ngspice. 

ngspice 42 and ngspice 42plus created "new" issues that ngspice 38 didn't have. I'm retesting simulations from a year ago to see why ngspice 38 "works" and ngspice 42/42plus doesn't. Could be my original circuits should never have worked under ngspice 38...

tomhajjar, 2024-01-24:
I forgot to give an update...

The new library works 100% with the "latest" (as of a week ago) nightly build ngspice "42plus". Even using ngspice 42, the majority of devices in the library worked. 

I assume ngspice 43 will be released soon.

ra3xdh, 2024-01-29:
I have uploaded a new library into the repository. Closing as completed.

closed 2024-01-29 by ra3xdh