ra3xdh/qucs_s #437: Paintings, nets or wires with Dimension of "0"

tomhajjar, 2024-01-07:
As outlined in #426:

Make a wire or any "Painting" like a line, ellipse, circle..etc that is so short or small to be unviewable and place it out of the desired viewable area.

Attached is a schematic with a line and ellipse with a dimension of "0". If these end up "off screen" they both screw up "Zoom All" AND they cannot be selected for deletion. Only a text editor will find and delete them.

Shouldn't the ability to make items with dimensions of "0" be removed.

[Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/13854249/Test_prj.zip)

ra3xdh, 2024-01-08:
Yes, the best solution would be to disable zero dimensions everywhere.

wawuwo, 2024-01-08:
I think I'll make a patch today

tomhajjar, 2024-01-08:
"adds 1 to the element's bounding rect"

I assume this doesn't actually alter the dimension

ra3xdh, 2024-01-08:
The best solution would be to add the check in `MouseResizeMoving()` method in classes derived from `Painting`. This check should not allows to create zero size object and set the minimal size to 1x1.

wawuwo, 2024-01-08:
@tomhajjar I've made a patch and it seems to work nice, at least with a file you attached to this issue.
[image]

tomhajjar, 2024-01-08:
No wire, painting or object of size "0" should be allowed

wawuwo, 2024-01-08:
@tomhajjar do you have any other "problematic" files to test the patch? Alternatively, you can build from the patched branch and test yourself :) If it's OK I'll open a PR

tomhajjar, 2024-01-08:
I can't compile Qucs so I wait for Vadim to create a Windows version...

wawuwo, 2024-01-09:
OK, I'm opening a PR then

closed 2024-01-14 by ra3xdh