ra3xdh/qucs_s #438: Proper Voltage Source Format

tomhajjar, 2024-01-07:
Can you explain why one format of input V "works" and the other doesn't. 

Isn't redundant to use "V=DC 0 AC 1 0 sin(0 1 796k 0 0 0)" versus "V=sin(0 1 796k 0 0 0)"?

[image]

[image]

[Matching_Network.zip](https://github.com/ra3xdh/qucs_s/files/13854616/Matching_Network.zip)

ra3xdh, 2024-01-08:
> Can you explain why one format of input V "works" and the other doesn't.

It is not a Qucs-S bug. You have wrong SPICE syntax. The AC level is not defined and Ngspice assumes it is zero. And you obtain division by zero error.  TRAN simulation works as expected with this source. The AC level should be always defined explicitly if you are using full SPICE syntax. The blue source defines AC level automatically, but your should define everything by yourself if you decided to use the red source.

closed 2024-01-08 by ra3xdh

tomhajjar, 2024-01-08:
I'm confused. 

How does my input string "V=sin(0 1 796k 0 0 0)" not define the voltage input as DC offset=0, AC Amplitude=1, F=796k...etc? The string matches the manual as shown below.

How is "V=DC 0 AC 1 0 sin(0 1 796k 0 0 0)" not redundant in specifying DC=0 and AC=1 twice?

[image]

tomhajjar, 2024-01-08:
I Grep'd the ngspice examples folder and found many variations of Voltage source syntax including the one I used.

ra3xdh, 2024-01-08:
There is no Qucs-S error here. The `SIN` defines the model for TRAN only. AC level and TRAN level may be different. Try the following simulation in CLI and you will see that your source produces only zero output on AC. Here is the screenshot. Probably it is Ngspice bug or manual may be read misleading here. 
[image]

tomhajjar, 2024-01-08:
Changing the string to V=DC 0 AC 1 also works for AC analysis. It was a annoying using such a long string.

[image]